ckb-next/ckb-next #686: Keyboard and mouse lights turn off completely and are unresponsive after ckb-next-daemon starts

Redempt, 2020-12-29:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information

### Source
How and where did you get this program, e.g.:
  * Package from AUR/PPA/COPR/...
  * Built manually using quickinstall script
  * Pre-built dmg for macOS

Affects aur/ckb-next-git, aur/ckb-next

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather
[ckb-next-report-1609268029.gz](https://github.com/ckb-next/ckb-next/files/5751812/ckb-next-report-1609268029.gz)

## What is the program's current behaviour?
After starting ckb-next-daemon, the keyboard and mouse lights turn off entirely regardless of what profile is selected and regardless of the idle timeout (which is when this would be expected)

## What is the expected behaviour of the program?
The keyboard lights should stay on when ckb-next-daemon is started

## Any other notes & comments?
Thank you for creating this amazing tool!

Espionage724, 2020-12-30:
I'll need to generate a log report, but I'm having a similar issue that affects my mouse (HARPOON RGB). When 
ckb-next-daemon starts, my mouse DPI flashes on and off about every second, and doesn't work (moving the mouse doesn't move the cursor).

If I stop ckb-next-daemon, the mouse works fine, but then my keyboard stops working entirely (K70 RGB RAPIDFIRE).

Neither of these happened when I last used ckb-next. I run openSUSE Tumbleweed, and it looks like they updated their package to 0.4.3. Looking at GitHub, this release happened about 2 weeks ago, and that time-frame is about when these issues started happening, so it seems like it's something with the newest 0.4.3 version causing this.

Redempt, 2021-02-27:
Turns out I'm a complete idiot. I pressed the brightness button on my keyboard by accident, which turned the lights off, and only now, almost 2 months later, did I realize that's what happened. So there's a button on this keyboard that cycles through brightness settings for the backlight. I must have had it at the highest one, and hit that button by accident, which turned the lights off completely. I honestly forgot this button existed, but I just hit it by mistake again today and lo and behold, the lights came back.

closed 2021-02-27 by Redempt

BloodyIron, 2021-11-28:
For future humans finding this page. I had a recent power outage, like 30 seconds, and it might have caused the lighting level to be reconfigured on my mouse (speculation), since it was the only thing that seemed abnormal in the area. I had thought that my mouse's profile was completely wiped, as when the daemon loaded the mouse's lights would just instantly turn off (they are on during boot with the default profile). I checked the configuration in ckb-next and everything actually looked good, all dpi and colour settings looked the same.

Thanks to @Redempt 's post above, it inspired me to look for a brightness setting. Sure enough, it was set to 0% in ckb-next. Set it to 100% and the lights turn on, and so far seems to be working just how I want it, all settings still there (I think). So hope that helps someone in the future!

BloodyIron, 2022-01-09:
Okay the brightness on my mouse, M65 RGB Elite, keeps getting set to 0%, and the lights also can stay off even if I turn it back up to 100%. Sometimes when I turn the lights from 0% (which I did not set it to do) to 100%, the lights come back on. Sometimes, the lights do not come back on, until I unplug and re-plug the mouse back in, and the lights are on then. Unsure where the root cause of this is :(