ckb-next/ckb-next #687: RGB Lighting turns off when ckb-next is open [K68 RGB]

CoffeeBoi47, 2021-01-03:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information

I noticed that my keyboards rgb had turned off one day and check ckb-next and found that I could not change the rgb at all. I closed ckb-next and the rgb came up and the keyboard did the default rainbow animation. I then opened ckb-next again and the rgb shut off. I have rebuilt this multiple times to no avail. This program worked fine for a long time and I have no idea why it is now not functioning properly 

### Source

  * Built manually using quickinstall script

## Logs & crash reports
[ckb-next-report-1609689519.gz](https://github.com/ckb-next/ckb-next/files/5761826/ckb-next-report-1609689519.gz)

## What is the program's current behaviour?

Keyboard rgb lighting doesn't work if ckb-next is open

## What is the expected behavior of the program?

Keyboard rgb works and is controllable while ckb-next is open

## Any other notes & comments?

I have a Corsair K68 RGB keyboard

stevef9432203, 2021-01-12:
Same exact problem... Strafe RGB keyboard.  Fedora 32. If ckb-next loaded Leds are off. Ckb-next thinks the Leds are active. Genuine Pita problem

stevef9432203, 2021-01-12:
I Just downgraded the Ckb-Next restarted and hit the * key next to the Symbol and back alive once more

maetthew, 2021-01-12:
I had the same problem, started happening randomly yesterday. Fedora 33. First I thought ckb-next had updated (i did `sudo dnf upgrade` yesterday) but `dnf history list ckb-next` showed it hadn't updated sincce 3rd of january. However, I did as @stevef9432203 mentioned; downgraded, rebooted and hit the * key and it came back.

stevef9432203, 2021-01-15:
Glad I Could help

Stevef9432203

J37T3R, 2021-01-23:
I have a similar (possibly related?) issue with my K70 and Scimitar where on start ckb-next turns my RGB off, but in settings thinks it's on. Changing the light level puts things back to normal.

KaffeinatedKat, 2021-01-24:
> I have a similar (possibly related?) issue with my K70 and Scimitar where on start ckb-next turns my RGB off, but in settings thinks it's on. Changing the light level puts things back to normal.

That was it, that you so much. The brightness was just at 0%

closed 2021-01-24 by KaffeinatedKat