ckb-next/ckb-next #690: Documentation update for Ubuntu 20.04: Qt5LinguistToolsConfig.cmake not found

cj2001, 2021-01-05:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information

The `README.md` cuts off at Ubuntu 16.04 where a simple `./quickinstall` work just fine.  However, there is a change in Qt5LinguistTools for 20.04 (and I think 18.04) as described in [this StackOverflow post](https://stackoverflow.com/questions/51698075/cmake-cannot-find-qt5linguisttools-in-docker-ubuntu-18-04) whereby you need to also do `sudo apt install qttools5-dev`.

### Source
How and where did you get this program, e.g.:
  ~~* Package from AUR/PPA/COPR/...~~
  * Built manually using quickinstall script
  ~~* Pre-built dmg for macOS~~

## Logs & crash reports

Upon issuing `./quickinstall` in Ubuntu 20.04, I get:

```
CMake Error at src/gui/CMakeLists.txt:363 (find_package):
  Could not find a package configuration file provided by "Qt5LinguistTools"
  with any of the following names:

    Qt5LinguistToolsConfig.cmake
    qt5linguisttools-config.cmake

  Add the installation prefix of "Qt5LinguistTools" to CMAKE_PREFIX_PATH or
  set "Qt5LinguistTools_DIR" to a directory containing one of the above
  files.  If "Qt5LinguistTools" provides a separate development package or
  SDK, be sure it has been installed.
```

## What is the program's current behaviour?

## What is the expected behaviour of the program?

## Any other notes & comments?

This is a simple documentation fix in the `README.md` where you just need to add an additional blurb for Ubuntu 18.04 and 20.04 adding `sudo apt install qttools5-dev` prior to installation.  I am happy to put in that very small PR if you are open to outside contributors, or it is a very minor thing on your own.

hchen98, 2021-06-20:
To fix this issue, you can try to install qttools5-dev via:
`sudo apt install qttools5-dev -y`

My problem was solved after installing this. 

[Reference](https://stackoverflow.com/questions/51698075/cmake-cannot-find-qt5linguisttools-in-docker-ubuntu-18-04)

closed 2021-06-20 by tatokis

ssainz, 2021-10-03:
what worked for me: 
```
sudo apt install qttools5-dev-tools -y
```