ckb-next/ckb-next #693: K70 RapidFire: Back from sleep to a dead daemon

arnauddupuis, 2021-01-18:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information

### Source
How and where did you get this program: Fedora 33 repository

sudo dnf info ckb-next

Last metadata expiration check: 1:58:06 ago on dim. 17 janv. 2021 14:12:04.
Installed Packages
Name         : ckb-next
Version      : 0.4.3
Release      : 1.fc33
Architecture : x86_64
Size         : 2.6 M
Source       : ckb-next-0.4.3-1.fc33.src.rpm
Repository   : @System
From repo    : updates
Summary      : Unofficial driver for Corsair RGB keyboards
URL          : https://github.com/ckb-next/ckb-next

## Logs & crash reports

[ckb-next-report-1609716743.gz](https://github.com/ckb-next/ckb-next/files/5827080/ckb-next-report-1609716743.gz)

## What is the program's current behaviour?

When the computer wakes up from sleep, very often the ckb-next daemon's process turns into a zombie process. The keyboard is unusable and most of the time unplugging/plugging the keyboard does not help. Actually plugging any USB keyboard seems to be impossible.
I reproduce that behavior so much that I deactivated the daemon for now.

## What is the expected behaviour of the program?

Resume normal operation when coming back from sleep/hibernation.

## Any other notes & comments?

Thank you so much for your work!

tatokis, 2021-01-18:
Can you please generate a backtrace when this issue is occurring?
You'll need the debug symbols for the package you installed. Dnf should be able to install them.
Once they're installed, make sure you have another keyboard plugged in and when this issue happens,
```
sudo gdb attach `pidof ckb-next-daemon`
```
Then in the gdb prompt
```
set pagination off
thread apply all bt full
```
and copy paste the output here.

I feel like this is a kernel bug due to the fact that you can't sigkill the process.

arnauddupuis, 2021-01-20:
Sure. I'll look into it over the weekend and post the results.

dmocek, 2021-01-26:
I have a problem similar to this that showed up in 0.4.3 (this issue does not happen for me with v0.4.2).  When I wake my laptop, my keyboard does work, but I see the default rainbow colors even though I have custom colors set.  To get my custom colors back, I need to stop ckb-next and pkill ckb-next-daemon, then restart the daemon and re-start ckb-next.  I'm running Ubuntu 20.04.

dmocek, 2021-01-27:
I found that if I wake my computer before I plug the keyboard in, I have to go through the above steps to get it to recognize the keyboard again.  If I plug my keyboard in before waking the computer, everything works OK.  v0.4.2 recognized my keyboard even if I plugged it in after waking the computer.

dmocek, 2021-01-29:
One more thing...if I leave the keyboard plugged into my computer and I Suspend then wake it, I don't have to re-start the daemon, but I do have to restart the app to get it to recognize my keyboard and apply my settings.

tatokis, 2021-01-30:
Try the latest code from master.
This should improve the situation https://github.com/ckb-next/ckb-next/commit/e48e6e6db4b905e2f307e47e7daf2aedb02b9ead

dmocek, 2021-02-03:
This fixes the issues I'm seeing.  Thanks tatokis!

dmocek, 2021-02-14:
This fix seemed to take care of the issue, but it's back, though not as bad.  I'm running version ckb-next 0.4.3-7-ge48e6e6.  When I wake my machines I still get the state where ckb isn't running.  I have to pkill both the client and daemon which I can now do.  When I re-start the client, it's working again (the daemon re-starts itself).  I am seeing a ton of these in the logs:

Feb 13 08:33:26 gp18 ckb-next-daemon[4532]: [I] Attempting reset...
Feb 13 08:33:26 gp18 ckb-next-daemon[4532]: [E] os_resetusb (via usb.c:614): resetusb failed: No such device
Feb 13 08:33:26 gp18 ckb-next-daemon[4532]: [E] usb_tryreset (usb.c:622): Reset failed. Disconnecting.
Feb 13 08:33:26 gp18 ckb-next-daemon[4532]: [E] reactivate_devices (usb.c:930): vt->active(kb, 0, 0, 0, 0) failed after resume
Feb 13 08:33:26 gp18 ckb-next-daemon[4532]: [E] os_usbsend (via device_keyboard.c:40): No such device, res = 0xffffffff

tatokis, 2021-02-16:
https://github.com/ckb-next/ckb-next/commit/908dcfd2cafb995daaccbbff964e8fb6ac0bdbb9
This should fix that as well

dmocek, 2021-05-19:
Issue has been fixed for me.  Thanks!

fritz-fritz, 2025-04-09:
Just experienced this general behavior myself. I know this is a stale issue that probably should have been closed long ago... But it might actually be related so filing here!

Keyboard: K70 RGB Pro
Mouse: Harpoon Wireless RGB
CKB: forked from 0.6.2 with patches for wl-scroll and DPI

Came back home and woke computer that was sleeping.
Keyboard and mouse were completely non-responsive.
Unplugged and plugged in USB devices to resume functionality yet not detected by CKB.
`sudo systemctl restart ckb-next-daemon.service` hung.
`sudo pkill ckb-next-daemon` failed.
`sudo kill -9 $(pidof ckb-next-daemon)` allowed the restart to complete.
Devices detected by CKB.

[ckb-next-report-1744160126.gz](https://github.com/user-attachments/files/19658701/ckb-next-report-1744160126.gz)

Of note is I just upgraded my kernel this morning to 6.12.20-amd64.

If I can replicate the issue I will attempt a backtrace per your directive above.