ckb-next/ckb-next #697: K60 series keyboard support

TheFortunato, 2021-02-03:
- Check if your device is listed in the [wiki](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware). (Done)
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues). (Done)
- If possible, try the latest code by building the software manually. (Did not do, don't know really how exactly)

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

K60 RGB PRO SE

[ckb-next-report-1612386618.gz](https://github.com/ckb-next/ckb-next/files/5921500/ckb-next-report-1612386618.gz)

SharkEzz, 2021-02-12:
`If possible, try the latest code by building the software manually. (Did not do, don't know really how exactly)`
I've just tried building CKB from the latest commit and I can confirm that it did not work.

thoffmeyer0, 2021-03-03:
Would also be willing to test patches for this keyboard.

zerofill, 2021-04-14:
Same issue with K60 RGB Pro
[ckb-next-report-1618436159.zip](https://github.com/ckb-next/ckb-next/files/6313839/ckb-next-report-1618436159.zip)

zerofill, 2021-04-14:
On a side note... I did use the latest cloned from the repo.

BoomSaag, 2021-04-15:
Yesterday I ordered the K60 Rgb Pro Se. I am still waiting for it to be shipped. Will also assist in testing once support is added. 
I am holding thumbs that support will be added soon.

voondo, 2021-05-06:
Here is a report for CORSAIR K60 RGB PRO Low Profile
[ckb-next-report-1620296242.gz](https://github.com/ckb-next/ckb-next/files/6433693/ckb-next-report-1620296242.gz)

Logan007, 2021-06-02:
I tried to implement K60 RGB LP (`...:1bad`) support analogous to #239 . As I have just not gotten it working at all, I wiresharked Windows VM-USB which did not show the typical `07` messages. Instead, it had USB payload starting with `08` a lot.

Could anyone confirm? Maybe a new protocol?

Logan007, 2021-06-06:
I have just verified that the captured packets starting with `08` seem to form the protocol for _K60 RGB LP_'s coloring. I actually replayed the packets using some C code which I have posted at [pastebin](https://pastebin.com/XYetM5iP).

The code contains all the captured packets (split into `reset[]` and `pattern0[]`, not all of them are used, note the loop counter `i` ranges), one copy of an RGB packet with manually applied changes (`pattern1`, to make it visible) as well as annotated findings on presumed packets' purpose.

May it help to integrate support for this keyboard into ckb-next!

tatokis, 2021-06-06:
The 08 is a bitfield. The 4th bit means "the device is to send a response". The first three bits control which device the packet needs to be sent to.
I see it uses 64 byte packets, so add the USB id as a k95 platinum xt in the k95-xt branch and you should have a starting point.

oliver602, 2021-08-12:
Another report in case it is useful. UK layout K60 RGB PRO Low Profile. Model RGP0122.
[ckb-next-report-1628773498.gz](https://github.com/ckb-next/ckb-next/files/6975671/ckb-next-report-1628773498.gz)

5ch4um1, 2022-03-13:
Just to give some positive feedback, K60 RGB Pro (:1ba0) here, ckb-next built from source on ubuntu 18.04, works like a charm. thank you very much!

joonapalaste, 2022-03-14:
> > Just to give some positive feedback, K60 RGB Pro (:1ba0) here, ckb-next built from source on ubuntu 18.04, works like a charm. thank you very much!
> 
> I have the K60 RGB Pro SE, but it does not detect it. Don't know if I am doing something wrong.
> 
> [ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/8246073/ckb-next-dev-detect-report.gz) s

Same here. I am running ckb-next 0.4.4-178-g967f440 built from the latest sources on GitHub on Fedora 35, and while ckb-next itself built and started just fine, it's still not recognising my Corsair K60 keyboard.

tatokis, 2022-03-15:
@BoomSaag You are not doing anything wrong. Thanks to your report, I have added the id for your keyboard https://github.com/ckb-next/ckb-next/commit/a5c5c4f1ce86ccb07dc0231d209e7c605d0e14c4. As it's obviously untested, make sure you have another keyboard pugged in.

@joonapalaste Assuming you have resolved the bios mode/firmware issue as per https://github.com/ckb-next/ckb-next/issues/722#issuecomment-986157437, please generate and attach a new report.

BoomSaag, 2022-03-15:
> @BoomSaag You are not doing anything wrong. Thanks to your report, I have added the id for your keyboard [a5c5c4f](https://github.com/ckb-next/ckb-next/commit/a5c5c4f1ce86ccb07dc0231d209e7c605d0e14c4). As it's obviously untested, make sure you have another keyboard pugged in.
>
@tatokis Thank you. I however do not fully understand your advice. You say I should have another keyboard plugged in and not my K60 Pro RGB SE since it is untested. Why is that? Are you afraid that it might malfunction?

tatokis, 2022-03-15:
@BoomSaag If the daemon does not understand the key packets your k60 is sending, or it doesn't know how to talk to the keyboard correctly, at best you'll get wrong key mappings, and at worst no keyboard input whatsoever while ckb-next-daemon is running. Having another keyboard plugged in means you can stop it to get the k60 working again without having to go into your OS' recovery mode to prevent ckb-next-daemon from autostarting (if you install it system wide).

Alternatively, you can build and run it without installing it and if input from the k60 stops working you can just power off/reboot with the power button on your computer.

It is extremely unlikely to happen in your case, but better safe than sorry.

BoomSaag, 2022-03-15:
@tatokis OK now I understand thank you. I do have a spare keyboard. I can also SSH from my phone and disable the daemon. 
Is there a way I can assist in testing the K60 Pro RGB SE so that you can categorize it as supported? I mean using it is obviously the first step, but how do I report back to you? Just start a new thread under issues then update there?

tatokis, 2022-03-15:
@BoomSaag if the keys work and the lights are mapped correctly, that's all you need to test.
This issue is fine.

BoomSaag, 2022-03-15:
@tatokis All seems to work 100%. I am loving the software. Thanks!

WhateverFits, 2022-03-22:
Works great on my Corsair CORSAIR K60 RGB PRO. Thanks for the commit!

Ravenslofty, 2022-05-29:
Supported as of #798, along with commit a5c5c4f1ce86ccb07dc0231d209e7c605d0e14c4 for the K60 PRO RGB SE, and 48173f13f8e992397f64f6075a6820aa09863809 for K60 PRO RGB Low-Profile.

closed 2022-05-29 by Ravenslofty