ckb-next/ckb-next #698: Corsair Dark Core RGB

mandraid, 2021-02-07:
Would it be possible to get this added to the list of supported devices?

Ravenslofty, 2021-02-07:
I think this is a duplicate of #101, unless Corsair have released another Dark Core.

mandraid, 2021-02-07:
I checked here: https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware but did not see it listed.

Ravenslofty, 2021-02-07:
Yes, that's because it's not supported. We have a patch to add support for it, but testers for it have come and gone often enough that it has sat idle for a while now.

thoffmeyer0, 2021-03-03:
I would be interested in giving feedback and such. I run Arch Linux, I also would request support for the k60 keyboard, also willing to test patches.

usrtrv, 2021-03-23:
@Ravenslofty Where's the patch? #101 mentioned the wireless branch but that didn't work for me.

Ravenslofty, 2021-03-23:
The patch *is* the wireless branch. If it doesn't work, we'd appreciate you joining us in IRC so we can get it a step closer to working.

medoix, 2021-05-12:
Happy to help test this out (i have a Dark Core RGB PRO SE) is this the same device being discussed here or an even newer model?
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/6466053/ckb-next-dev-detect-report.gz)

closed 2024-05-01 by gravityfargo