ckb-next/ckb-next #702: Not working/How to use the Mode Event Manager

LivingWithHippos, 2021-03-03:
## General information

I'm trying to use the switch to change mode on Manjaro but it does not seem to work.

I've seen there are 4 options:
- program path: checking from the task manager I've added /usr/lib/firefox/firefox. It does not work
- Window Title: depends on the current page so it's not useful
- Instance Name: I don't know what it is 
- Class Name: I get this with `xprop` checking `WM_CLASS(STRING) = "Navigator", "firefox"` or `WM_CLASS(STRING) = "jetbrains-studio", "jetbrains-studio"`. It ain't working (i also tried adding quotes)

I've tried permissive combinations such as "path start with /urs/lib/firefox OR Window Title contains Firefox OR Class Name contains firefox" but it's not working (any program, Firefox is just an example)

### Source
How and where did you get this program, e.g.:
  * Package from AUR

## Logs & crash reports

[ckb-next-report-1614814301.gz](https://github.com/ckb-next/ckb-next/files/6079824/ckb-next-report-1614814301.gz)

## What is the program's current behaviour?

Current/Active screen does not change the profile

## What is the expected behaviour of the program?

Changing program change the active mode

## Any other notes & comments?

The modes are in the same profile.

Yes, the events are enabled ;D

tatokis, 2021-03-07:
Do you get any errors if you run `ckb-next` from a terminal, after quitting the application? (Either through the quit button in settings or from the tray icon.)

LivingWithHippos, 2021-03-07:
It all started working, I don't know if I was missing a restart maybe?

closed 2021-03-07 by LivingWithHippos

ClemGlacee, 2021-09-12:
Instance Name what is that? and how do you get the class name?

ClemGlacee, 2021-09-12:
> It all started working, I don't know if I was missing a restart maybe?

I'm having this issue every time at first launch, I always need to quit and relaunch for the profiles to work... Have you found a work around?

LivingWithHippos, 2021-09-12:
Instance name is one of the ways to set the program aside from path etc. It's on a dropdown. I solved this issue by restarting the pc, I don't have it anymore. I also switched  mouse because it broke

ClemGlacee, 2021-09-12:
> Instance name is one of the ways to set the program aside from path 

Yes I know about that, but I can't find it for my program nor what it is exactly

Sadly restarting my computer doesn't change the issue, I still have to restart the program every time which is very very annoying.. Do you use some type of macro or profile still (with another program) or did you give up on that ?

LivingWithHippos, 2021-09-12:
It just started working for me after the reboot. I am now using a razer naga because I wanted to try another similar mouse. I suggest you to open another issue

ClemGlacee, 2021-09-12:
Yeah I probably will open a new one but it seems like the dev are kinda overwhelmed by the amount of issues
Anyway thank you for the help and I hope it's working fine with your new mouse :)

LivingWithHippos, 2021-09-12:
It's better in some ways and worse in others. It has the advantage to save macro in its inner memory so you can use it without software if you don't mind a single profile

ClemGlacee, 2021-09-12:
well at least you have macros, on ckb it's not very optimal imo