ckb-next/ckb-next #704: Corsair Scimitar RGB Elite doesn't work at all

RoyvanEmpel, 2021-03-20:
## General information
Hello! I just got the Corsair Scimitar RGB Elite, plugged it in and ckb-next picked it up in the software but the mouse doesn't work.
When I reboot the mouse works in the login screen and for a few seconds when logged in and after that the mouse stops working, No input from the buttons/Sensor.

I am running Pop!_OS 20.10.

##log
[ckb-next-report-1616242893.gz](https://github.com/ckb-next/ckb-next/files/6175526/ckb-next-report-1616242893.gz)

### Source
How and where did you get this program, e.g.:
Built manually using quickinstall script

## What is the program's current behaviour?
Starts and doesn't look like anything is going wrong, but for some reason the scimitar doesn't seem to work. It works for a few seconds at boot and thats it. The mouse is showing up in CKB-next

## What is the expected behaviour of the program?
It showing up like it's currently but i would expect the mouse to work.

## Any other notes & comments?
I did try rebooting/unplugging and reinstalling CKB-next but it doesn't seem like that changed anything.

tatokis, 2021-03-20:
Try plugging it on a different port. If using USB 3, try USB 2 and the opposite.
Avoid using any hubs, front USB ports or cable extensions.

Based on the logs, the mouse just doesn't respond to any messages ckb-next sends to it.

RoyvanEmpel, 2021-03-20:
It was already straight in my motherboard so i plugged it in my hub (didn't work) plugged it again in my motherboard. (didn't work) uuuuh replugged it again and it started working somehow?!

I guess it's working now, I should really have tried that -_-

closed 2021-03-20 by RoyvanEmpel

tatokis, 2021-03-20:
You can also try adding the usbcore quirks to the kernel command line to see if it helps as well.
`usbcore.quirks=1b1c:1b8b:gn`

https://wiki.ubuntu.com/Kernel/KernelBootParameters

RoyvanEmpel, 2021-03-20:
I used `sudo kernelstub -a usbcore.quirks=1b1c:1b8b:gn` to add it since pop doesn't use grub. Thanks for the help!

tatokis, 2021-03-20:
I did not know that, thank you!