ckb-next/ckb-next #705: K60 RGB PRO (CH) [1b1c:1ba0]

F-Hauri, 2021-03-21:
## CORSAIR K60 RGB PRO Mechanical Gaming Keyboard

Just tested today, with latest version.

This KBD with USB id: `1b1c:1ba0` seem not same keyboard than `#697` (`1b1c:1b8d`)

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

I've manually runned `./ckb-next-dev-detect` from disant ssh session. (desktop is not mine, I'm buildild install while user is working on;)

So the report will miss X related infos. I hope this is not harmful.

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/6177847/ckb-next-dev-detect-report.gz)

YeisonHunt, 2021-03-29:
OMG, this is to support the K60 RBG PRO SE?

F-Hauri, 2021-04-11:
SE is not CH_FR, but look like to be same...

khris190, 2021-06-09:
[ckb-next-report-1623221483.gz](https://github.com/ckb-next/ckb-next/files/6621660/ckb-next-report-1623221483.gz)
this is SE

jblievremont, 2021-07-29:
Note that French K60 RGB PRO (AZERTY layout, P/N: CH-910D019-FR) seems to have the same device ID (`1b1c:1ba0`) - see [ckb-next-report-1627594405.gz](https://github.com/ckb-next/ckb-next/files/6903781/ckb-next-report-1627594405.gz).

For someone with some knowledge of system programming and limited knowledge of C and C++, what could help move this forward? I have a Windows 10 box on which I can capture USB traffic if necessary.

lotar1954, 2021-10-01:
What is the status of K60 RGB PRO? Are there any news?

Ravenslofty, 2022-05-29:
Supported as of #798.

closed 2022-05-29 by Ravenslofty