ckb-next/ckb-next #706: Performance improvement

aaronw2, 2021-03-23:
## What is the program's current behavior?
From what I can tell, it looks like for animations ckb-next is executing a new process for each animation frame (at least with rain).
## What would you like it to do instead?
It would be far more efficient to have the animation process continue to run and just wait for input instead of terminating after each frame, only to start a new process for the next frame. The amount of processing to create a new process is non-trivial.
## Any other notes & comments?

tatokis, 2021-03-23:
That shouldn't be happening. How do you know it's creating a new process? Can you check if the animation is crashing?

Ravenslofty, 2021-03-23:
For example, [animscript.cpp](https://github.com/ckb-next/ckb-next/blob/master/src/gui/animscript.cpp) creates a single process and talks to it through pipes to receive the animation frame data. Since animations naturally have state, it would make no sense to design the animation system to spawn a process for each frame anyway.