ckb-next/ckb-next #709: Support for the new Sabre Pro & Sabre RGB Pro Championship Series mices

ale2695, 2021-04-24:
Hi there,
I saw that the new Sabre Pro Championship Series and Sabre RGB Pro Championship Series are not supported by ckb-next, could you work on their implementation?
I own the Sabre RGB Pro Championship Series since a few days and I would like to use it, but in the current stable version (0.4.4) it is not supported
Link on the Corsair website: 
- [Sabre RGB Pro Championship Series](https://www.corsair.com/us/en/Categories/Products/Gaming-Mice/MOBA-Strategy-%26-Action-Mice/SABRE-RGB-PRO-CHAMPION-SERIES-FPS-MOBA-Gaming-Mouse/p/CH-9303111-NA)
- [Sabre Pro Championship Series](https://www.corsair.com/it/it/Categorie/Prodotti/Mouse-gaming/Mouse-ad-azione-rapida-per-FPS/SABRE-RGB-CHAMPION-SERIES-FPS-MOBA-Gaming-Mouse/p/CH-9303101-EU)
Below the report generated by my installation of ckb-next on Arch Linux
[ckb-next-report-1619279540.gz](https://github.com/ckb-next/ckb-next/files/6370071/ckb-next-report-1619279540.gz)

tatokis, 2021-04-24:
I won't have time to look into this for a while, but you can try getting the source code from the [k95-xt](https://github.com/ckb-next/ckb-next/tree/k95-xt) branch, and changing the usb id in `src/device/daemon/usb.h` for the ironclaw wireless rgb (P_IRONCLAW_W_U) to yours.

Just be prepared if something goes wrong, your mouse might stop working while ckb-next-daemon is running.
Even though I'd advise installing it system wide before testing, you'd need to manually stop it with `sudo systemctl stop ckb-next-daemon` and/or uninstall it.

If it works and there are minor issues, we could probably fix them quickly.
Do send any logs from the daemon as well.

ale2695, 2021-04-25:
Hi tatokis,
thank you for your suggest, but sadly it doesn't work.
ckb doesn't show the device, and the mouse just act strange, it doesn't move the cursor when it move, but it simply act strange
Below you can find a screen recording of the problem:
[Video](https://drive.google.com/file/d/1IIEJKh7sugaShq1ixCk26czVVyRRCwJC/view?usp=sharing)
On the final the cursor works correctly as I used my MX Master 3 that obviously is not influenced by ckb
And there is the report generated by ckb-next:
[ckb-next-report-1619345190.gz](https://github.com/ckb-next/ckb-next/files/6371708/ckb-next-report-1619345190.gz)
Thank you so much!

Mario-119, 2022-04-11:
Bumping this, support would be nice for this mouse. Seems the side buttons don't work by default so it'd definitely help if that could be implemented as well.

Niklasw99, 2022-05-28:
Were this forgotten?

Niklasw99, 2022-10-11:
> Also, these mice (or at least Sabre RGB Pro Wireless Championship Series)
> support 2000Hz polling. I have no idea how it's achieved under Windows,
> given that under Linux it connects as a USB1 Full-Speed (12Mbps) device
> with 1ms minimum polling.
>
> Maybe there's some command switching it to USB2 mode (with minimum polling
> of 125us), or maybe it's a fake thing even under Windows, where it'd be SW
> which fakes the polling by using some interpolation or some other technique.
>
> According to the windows software it the polling can run up as high as
> 8000 Hz. Setting it higher than 1000 Hz will issue a warning about
> increased resource requirements. 8000 Hz requires a very beafy CPU
> according to the software's UI.
>
>
> <https://camo.githubusercontent.com/64c3266b431e37264fa063a4d63804f6620d5c0e38a5de16e87872a4359e5410/68747470733a2f2f692e696d6775722e636f6d2f38677a6d6265372e706e67>
>
> I'm not sure if sub-millisecond pulling rates are a problem in Linux. I
> wouldn't mind a missing support for higher pulling rates as long as we get
> all the other stuff (especially RGB) of the device supported in cbk-next.

Yeah and the dpi settings.

jinkhya, 2023-11-19:
Hello, 
Any chance those could be added ? :)
Thanks in advance,
Jin

freak2fast4u, 2024-04-18:
Is it possible I would benefit from this being implemented ? 
It seems I have the wireless variant in the same product line : 
https://www.corsair.com/us/en/p/gaming-mouse/ch-9313211-na/sabre-rgb-pro-wireless-champion-series-ultra-lightweight-fps-moba-gaming-mouse-ch-9313211-na

My mouse reports like this over usb cable : 
`Bus 001 Device 011: ID 1b1c:1b98 Corsair CORSAIR SABRE RGB PRO WIRELESS Gaming Mouse`
And the dongle reports like this : 
`Bus 001 Device 008: ID 1b1c:1ba6 Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver`

ckb-next won't pick-up on the mouse nor the dongle at time of writing.
I'm using version ckb-next version 0.6.0 from the official ppa on linux mint 21.3 virginia (based on ubuntu jammy).

gravityfargo, 2024-05-02:
> Is it possible I would benefit from this being implemented ? It seems I have the wireless variant in the same product line : https://www.corsair.com/us/en/p/gaming-mouse/ch-9313211-na/sabre-rgb-pro-wireless-champion-series-ultra-lightweight-fps-moba-gaming-mouse-ch-9313211-na
> 
> My mouse reports like this over usb cable : `Bus 001 Device 011: ID 1b1c:1b98 Corsair CORSAIR SABRE RGB PRO WIRELESS Gaming Mouse` And the dongle reports like this : `Bus 001 Device 008: ID 1b1c:1ba6 Corsair CORSAIR SLIPSTREAM WIRELESS USB Receiver`
> 
> ckb-next won't pick-up on the mouse nor the dongle at time of writing. I'm using version ckb-next version 0.6.0 from the official ppa on linux mint 21.3 virginia (based on ubuntu jammy).
> 
> [ckb-next-report-1713451269.gz](https://github.com/ckb-next/ckb-next/files/15026504/ckb-next-report-1713451269.gz)

Please open a seperate issue, that is a different device.

cotten321, 2025-12-25:
> Hi tatokis, thank you for your suggest, but sadly it doesn't work. ckb doesn't show the device, and the mouse just act strange, it doesn't move the cursor when it move, but it simply act strange Below you can find a screen recording of the problem: [Video](https://drive.google.com/file/d/1IIEJKh7sugaShq1ixCk26czVVyRRCwJC/view?usp=sharing) In the final the cursor works correctly because I used my MX Master 3 that obviously is not influenced by ckb And there is the report generated by ckb-next: [ckb-next-report-1619345190.gz](https://github.com/ckb-next/ckb-next/files/6371708/ckb-next-report-1619345190.gz) Thank you so much!

I also tried something similar to this and I could not get it to work. The mouse acts weird and then disconnects every minute or so.

anat0lius, 2026-01-14:
If you only want to **fix side buttons**, use this https://github.com/sezanzeb/input-remapper
Is a bit non-intuitive how to use it, so let me explain:
 
When you open the app, you will something like this:
[image]
I don't which of Corsair devices I'm supposed to select, but in my case with the first one it worked. Select it, then click on `new preset` . 
Then in Editor
- you first need to click on the `+Add` button of the Input side
- then click on `Record`
- click one of your side buttons. Do the same for the other side button. They will be represented as 1 and 2
- in the Output side, for each of the entries of the Input side, set Type `Key or Macro` and Target `mouse`
- in the text field, write down `BTN_EXTRA` for the side button 1, and `BTN_SIDE` for side button 2
- Finally click on `Apply` and test it

[image]

And that's it, from now on videogames can recognise the side buttons correctly (instead as of right/left mouse buttons).

DrewGitsIt, 2026-08-15:
@tatokis I have a proposed fix here: https://github.com/ckb-next/ckb-next/pull/1343

I realize it's a little late but I'm new to the game. If you get time, please review :)

I don't seem to have permissions to link that PR to this issue.