ckb-next/ckb-next #711: Branch k95-xt numpad issue

duonghao314, 2021-04-29:
When I install Ckb in branch `k95-xt` my numpad zone got an issue. 
Numpad 7 -> Nothing do 
Numpad 8 -> Numpad 7
Numpad 9 -> Numpad 8
Numpad 0 -> Numpad 9
Numpad .  ->  Numpad 0

Is this error can be corrected manually. I am currently configuring with macros, but I cannot handle numpad 7 key because it not response any key.

TheWebMachine, 2021-05-03:
I, too, have this issue. Just compiled from the k95-xt branch today. Here's my diddly-do:
[ckb-next-report-1620015354.txt](https://github.com/ckb-next/ckb-next/files/6412823/ckb-next-report-1620015354.txt)
(NOTE: The service crash is related to trying to change the polling rate in the GUI, for which I will open a separate bug, but this numpad issue also plagues me.)

TheWebMachine, 2021-05-03:
Found another mis-mapped key. The Menu modifier key is mapping as `<` Unbinding does nothing as it isn't the right "scancode" when you press the key

TheWebMachine, 2021-05-09:
So, it looks like the only key left to sort out is numpad 7. Everything else is working great as of 12f582d

TheWebMachine, 2021-06-17:
Has any progress been made on numpad 7 and the Menu modifier? (I misspoke in my last comment that only numpad 7 was still broken; Menu modi is still coding as `<` and can't be re-mapped.) Anything I can do to help? 

I've taken to scripting a stop of the ckb service automatically when my IDEs are launched so I can write without losing my sanity over that stupid 7 key...never realized just how much I rely on numpad when I code. 🤣️ So, yeah, if there's anything I can do to help, I'm all ears!

tatokis, 2021-06-17:
I've been reworking a big chunk of the usb code in order to merge the k95-xt branch to master under the experimental device flag, as they had diverged a lot, and the hacks were piling up. I'll get to this after I've finished merging and cleaning up.

FryDay, 2021-09-13:
Any update on this @tatokis? No 7 key gets annoying after a bit.

tatokis, 2021-09-27:
@FryDay not really, sorry. I did receive a K57 which behaves the same way in wired mode, so once I get some basic support for it, I'll fix the numpad 7 issue.

tatokis, 2021-09-30:
https://github.com/ckb-next/ckb-next/issues/543#issuecomment-931025424
Please test again after reading the above.