ckb-next/ckb-next #712: Wired Harpoon issues

djijitsu, 2021-04-29:
I wanted to be able to turn off the RGB on my wired Harpoon mouse. I installed ckb-next using terminal in Linux Mint Ulyssa and the mouse immediately stopped functioning. If I recall correctly I tried uninstalling the PPA, but I only gained control of some of the mouse buttons, but not the cursor. I tried reinstalling and brought up the ckb manager that says 'no device connected'.
I also think there were some interference with the laptop touch pad upon installation, but thankfully it got function back after a reboot.

Would appreciate some advice on how I should proceed to get a functioning mouse ASAP!

[ckb-next-report-1619713723.gz](https://github.com/ckb-next/ckb-next/files/6400235/ckb-next-report-1619713723.gz)

Ravenslofty, 2021-04-29:
Please join us in IRC, you can find the freenode button in the readme.

tatokis, 2021-04-29:
Your mouse is in bootloader mode. You either need to reinstall the firmware through ckb if possible, or plug it in to a computer with corsair's software to recover it.

djijitsu, 2021-04-29:
> Please join us in IRC, you can find the freenode button in the readme.

okay, thanks.

> Your mouse is in bootloader mode. You either need to reinstall the firmware through ckb if possible, or plug it in to a computer with corsair's software to recover it.

Thank you! Is there any way to reinstall the firmware manually through ckb?

djijitsu, 2021-04-29:
I got great help from the IRC channel! Thanks alot!

closed 2021-05-21 by Ravenslofty