ckb-next/ckb-next #716: Corsair Dark Core PRO SE RGB

medoix, 2021-05-12:
This device is a Bragi wireless device and is not currently supported even on the wireless branch.

Raising this to track the device for any potential future improvements or patch that may get this working in linux.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/6466194/ckb-next-dev-detect-report.gz)

Clofriwen, 2021-07-01:
Hello,

I joint my log too (wired & wireless) :
1b1c:1b7f Corsair General Purpose USB Hub -> Wireless dongle
1b1c:1b7e Corsair CORSAIR DARK CORE RGB PRO SE Gaming Mouse
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/6751077/ckb-next-dev-detect-report.gz)

Other Corsair device :
1b1c:0c10 Corsair Commander PRO
1b1c:1b20 Corsair Corsair STRAFE RGB Gaming Keyboard

jmiahman, 2021-12-23:
I am able to see settings.. I think
[image]
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/7771328/ckb-next-dev-detect-report.gz)

rmrfforewardslashstar, 2022-01-04:
Duplicate of #698 "Corsair Dark Core RGB"

T-bond, 2022-01-04:
@rmrfforewardslashstar Dark Core RGB is not the same device as the Dark Core RGB PRO (SE)

Dark Core RGB:
https://www.corsair.com/us/en/Categories/Products/Gaming-Mice/DARK-CORE-RGB-Performance-Wired-Wireless-Gaming-Mouse/p/CH-9315211-NA

Dark Core RGB PRO:
https://www.corsair.com/us/en/Categories/Products/Gaming-Mice/Wireless-Gaming-Mice/DARK-CORE-RGB-PRO-Wireless-Gaming-Mouse/p/CH-9315411-NA

Dark Core RGB PRO SE:
https://www.corsair.com/us/en/Categories/Products/Gaming-Mice/Wireless-Gaming-Mice/DARK-CORE-RGB-PRO-SE-Wireless-Gaming-Mouse/p/CH-9315511-NA

Difference between Dark Core RGB PRO vs the SE variant is only that the SE is Qi chargeable. I think that part is not important in this project. But the PRO vs non PRO is different in other hardware parts, like programable button count 8 vs 9.

NRG-R9T, 2022-02-09:
How's the progress with this? I use the same mouse (Corsair Dark Core PRO SE RGB) and although I was using it with Linux since 1,5 years. Somehow it did work wirelessly - I don't remember how I did it. It died suddenly and I got a new one in exchange swiftly. But now I can't figure out how to enable the slipstream wireless mode again on Arch Linux. I was on Ubuntu when I got the Mouse. 

The --enable-experimental flag also did not work, because "./bin/ckb-next-daemon" command didn't work.

Link to [Report](https://gist.github.com/NRG-R9T/fec50a4f12161dba83c794aa10174060) on Git

jmiahman, 2022-05-13:
Adding my report in case it can be a help.
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/8691059/ckb-next-dev-detect-report.gz)

gilvbp, 2022-05-27:
Someone willing to help?

Ravenslofty, 2022-05-29:
The Pro SE Wireless has been experimentally supported since #728, but we would need somebody to come join us in IRC so we can iron out issues (and please be patient while waiting for somebody to respond).

We've always been reliant on community testers for hardware, because we can't afford to purchase all the hardware to test things ourselves.

Ravenslofty, 2022-05-29:
We will still need you to come into IRC: a log file isn't enough by itself.

Clofriwen, 2022-05-29:
I'm in actually

gilvbp, 2022-05-30:
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/8794630/ckb-next-dev-detect-report.gz)

gilvbp, 2022-05-30:
[Corsair Dark Core PRO RGB wirelless]
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/8794639/ckb-next-dev-detect-report.gz)

Clofriwen, 2022-05-30:
Testing version, (only daemon changed)

That should support "Dark Core RGB Pro" &  "Dark Core RGB Pro SE", with experimental flag.

Both in Wired & Wireless mode.

GUI can show (normaly) : battery stat/level, Firmware, Bootloader Serial Number, Poll rate, and 4(+3 dpi) led on 9(+3) working.

[ckb-next-0.5.0-DarkCorePro.tar.gz](https://github.com/ckb-next/ckb-next/files/8794648/ckb-next-0.5.0-DarkCorePro.tar.gz)

Clofriwen, 2022-05-30:
I miss 2 lines : in src/daemon/led_bragi.c
    LED_CASE_M(P_DARK_CORE_RGB_PRO, 12);
    LED_CASE_M(P_DARK_CORE_RGB_PRO_WL, 12);

that must be between this lines (l.36) :
    LED_CASE_M(P_M55_RGB_PRO, 2);
    LED_CASE_M(P_DARK_CORE_RGB_PRO_SE, 12);

gilvbp, 2022-05-30:
@Clofriwen I noticed and fixed it, but thank you very muc
[ckb-next-0.5.0.zip](https://github.com/ckb-next/ckb-next/files/8794926/ckb-next-0.5.0.zip)
h

Clofriwen, 2022-05-30:
All done, "Corsair Dark Core RGB Pro SE" are functional now.
"Corsair Dark Core RGB Pro" should be too.

For my personal usage, i have removed "DARK_CORE_RGB_PRO" from "IS_EXPERIMENTAL" (in src/daemon/usc.h).
This modification isn't in this next archive.

[ckb-next-0.5.0-Dark_Core_RGB_Pro.tar.gz](https://github.com/ckb-next/ckb-next/files/8795581/ckb-next-0.5.0-Dark_Core_RGB_Pro.tar.gz)

This archive content only files was added/changed
[Dark_Core_RGB_Pro-Patch.tar.gz](https://github.com/ckb-next/ckb-next/files/8795584/Dark_Core_RGB_Pro-Patch.tar.gz)

Good tests and enjoy

gilvbp, 2022-05-30:
> All done, "Corsair Dark Core RGB Pro SE" are functional now. "Corsair Dark Core RGB Pro" should be too.
> 
> For my personal usage, i have removed "DARK_CORE_RGB_PRO" from "IS_EXPERIMENTAL" (in src/daemon/usc.h). This modification isn't in this next archive.
> 
> You can need to uninstall previous version before install this one. Button "Profile Switch" don't work, i wasn't able to find for map it.
> 
> [ckb-next-0.5.0-Dark_Core_RGB_Pro.tar.gz](https://github.com/ckb-next/ckb-next/files/8795581/ckb-next-0.5.0-Dark_Core_RGB_Pro.tar.gz)
> 
> This archive content only files was added/changed [Dark_Core_RGB_Pro-Patch.tar.gz](https://github.com/ckb-next/ckb-next/files/8795584/Dark_Core_RGB_Pro-Patch.tar.gz)
> 
> Thanks to Lofty, KitsuWhooa and Gil for help.
> 
> Good tests and enjoy

Great job!  I will create a fork from the original and pull request these modifications. Thank you so much!

Clofriwen, 2022-05-31:
Hello,
while checking the commit, I see 3 line I have forgotten to remove :
[image]

also, this things is not necessary :
[image]

Clofriwen, 2022-05-31:
Here the cleaned version
[ckb-next-0.5.0-Dark_Core_RGB_Pro.2.tar.gz](https://github.com/ckb-next/ckb-next/files/8807971/ckb-next-0.5.0-Dark_Core_RGB_Pro.2.tar.gz)
[Dark_Core_RGB_Pro-Patch.2.tar.gz](https://github.com/ckb-next/ckb-next/files/8807973/Dark_Core_RGB_Pro-Patch.2.tar.gz)

JoePfeiffer, 2022-06-19:
The battery charging/discharging indication appears to be reversed -- when my mouse is discharging it says "charging", and vice versa.

Clofriwen, 2022-07-07:
New pull opened at this place : https://github.com/ckb-next/ckb-next/pull/855
- that made from an current master version
- support an corrected battery status (improved compatibility with nxp protocole (Dark Core))

olokos, 2022-07-08:
Somewhere between 6 months and 1 year I tried the mouse on slightly less updated software and the mouse would output 1 and 2 for next page and previous page and feeling wonky, but that was earlier, right now I am on:

OS: Ubuntu 20.04.4 LTS x86_64 
Kernel: 5.13.0-52-generic
ckb-next 0.50

Mouse Corsair Dark Core RGB PRO SE

Features status:
Wireless:
Mouse movements are as expected
Previous/Next page button works as expected
Scroll up/down works as expected
Left/Right click works as expected
DPI UP/DOWN works as expected, using hardware defaults
Sniper button - works as expected, BUT instead of displaying a red light on the LED above the "sniper button" (below the scroll) it's displaying red light instead on the side DPI level indicator lights, so it kinda works as expected, but displays red on side instead of top leds.

Clofriwen, 2022-07-09:
Hello,

I have added some little thing (still in opened pull request https://github.com/ckb-next/ckb-next/pull/855) :
- DPI 3 level indicator & Sniper mode (working like in official)
- Battery indicator (under wheel) (green = charging, orange = low, red = critical) (like in official... but not sure for orange)
- Profile switch now working
- Some little cleaning

The DPI level working better is that in black color and not in animation
[image]

Grooty12, 2023-06-18:
Getting this weird image glitch

https://github.com/ckb-next/ckb-next/assets/85930101/af7e8ff1-bc7b-40f1-9261-dfdac7784bfb

Grooty12, 2023-06-19:
Sure thing
`
ckb-next 0.5.0-139-g2dcda6e9
Using "AMD" "AMD Radeon RX 6800 XT (navi21, LLVM 15.0.7, DRM 3.52, 6.3.8-arch1-1)"
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/rasmus/.config/ckb-next/ckb-next.conf"
Scanning "/usr/lib/ckb-next-animations/gradient"
Scanning "/usr/lib/ckb-next-animations/heat"
Scanning "/usr/lib/ckb-next-animations/invaders"
Scanning "/usr/lib/ckb-next-animations/life"
Scanning "/usr/lib/ckb-next-animations/mviz"
Scanning "/usr/lib/ckb-next-animations/pinwheel"
Scanning "/usr/lib/ckb-next-animations/pipe"
Scanning "/usr/lib/ckb-next-animations/rain"
Scanning "/usr/lib/ckb-next-animations/random"
Scanning "/usr/lib/ckb-next-animations/ripple"
Scanning "/usr/lib/ckb-next-animations/snake"
Scanning "/usr/lib/ckb-next-animations/wave"
QFont::setPixelSize: Pixel size <= 0 (0)
QFont::setPixelSize: Pixel size <= 0 (0)
QFont::setPixelSize: Pixel size <= 0 (0)
Starting "/usr/lib/ckb-next-animations/wave"
Starting "/usr/lib/ckb-next-animations/pinwheel"
Starting "/usr/lib/ckb-next-animations/wave"
Creating PulseAudio context
`
Running Kde on Arch Linux with Pipewire and Wayland if that helps

tatokis, 2023-06-19:
Does this bug happen if you also run it like this?
`LIBGL_ALWAYS_SOFTWARE=1 ckb-next`

Grooty12, 2023-06-19:
Yes, the same thing happens

tatokis, 2023-06-20:
@Grooty12 Can you please try again with the latest code from git?
It should be fixed by https://github.com/ckb-next/ckb-next/commit/b5acdeaeced7a4db24c87866d83a8807571388c9

Grooty12, 2023-06-20:
It's fixed yeah thanks

closed 2024-05-02 by gravityfargo