ckb-next/ckb-next #719: BSD support?

Indoors-Cat, 2021-05-31:
## What would you like it to do instead?
Any plans for a BSD version of this? I quite like this piece of software and would love to be able to use it on BSD as well as Linux.

Ravenslofty, 2021-05-31:
ckb-next does a reasonably good job of providing a backend to hide OS interfaces, so a port to a BSD is feasible. But I have been looking at the APIs they offer, and the choices range from 'very quirky' (FreeBSD with its glue shims) to 'why would you do that' (NetBSD would require ckb-next to effectively become a kernel driver) to 'not actually possible' (you can't inject input events from userspace in OpenBSD as far as I can tell).

So, uh, good luck.

Indoors-Cat, 2021-06-01:
Very interesting. I should've specified than I'm personally using FreeBSD, but didn't think there was that much variance between the different BSD's. I don't know much on coding (trying to learn but don't know very much as of now) so I really appreciate this response.

tdm4, 2022-04-17:
I would love to see BSD support as well. I run both FreeBSD and OpenBSD. From what it looks like in the source code, it looks like there's a hook into libusb. Whether or not the BSDs would allow these events injected in is another matter.

fred410, 2023-01-30:
yeah!! id love this to work on FreeBSD too!
there is something called "openrgb" it detects my K70 LUX RGB - and it says I can set colors only in direct mode (whatever that is) which is the whole keyboard or a specfic key. no profiles like "wave" or "breathing" possible.