ckb-next/ckb-next #720: K55 RGB PRO not detected

yochananmarqos, 2021-06-02:
### Source
Tried both AUR packages: `ckb-next` 0.4.4-1 and `ckb-next-git` 1:0.4.4.r6.g190ffeed-1

## Logs & crash reports

```
❯ ckb-next
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/yochanan/.config/ckb-next/ckb-next.conf"
Scanning "/usr/lib/ckb-next-animations/gradient"
Scanning "/usr/lib/ckb-next-animations/heat"
Scanning "/usr/lib/ckb-next-animations/invaders"
Scanning "/usr/lib/ckb-next-animations/life"
Scanning "/usr/lib/ckb-next-animations/mviz"
Scanning "/usr/lib/ckb-next-animations/pinwheel"
Scanning "/usr/lib/ckb-next-animations/pipe"
Scanning "/usr/lib/ckb-next-animations/rain"
Scanning "/usr/lib/ckb-next-animations/random"
Scanning "/usr/lib/ckb-next-animations/ripple"
Scanning "/usr/lib/ckb-next-animations/snake"
Scanning "/usr/lib/ckb-next-animations/wave"
Disconnecting from X
XCB EWMH Thread Quitting...
```

[ckb-next-report-1622673875.gz](https://github.com/ckb-next/ckb-next/files/6587912/ckb-next-report-1622673875.gz)

## What is the program's current behaviour?
No devices connected.

## What is the expected behaviour of the program?
It should detect it as support is claimed.

Synfoxic, 2021-07-05:
Is it known if the K55 RGB PRO Keyboard will be added? Just bought it today and was a little confused, as it did not detect it....
Although I could live with the fact, that I have no software for it. That's a price I am willing to pay for Linux :D 
but it would be really nice though, if it was added. Unfortunately I have absolutely no experience in this to do something myself :(

s3boun3t, 2021-07-18:
me too, i have a new corsair K55 rgb pro xt and it is not detected with ckb-next-git (0.4.4-77-g03d683c)

lsusb
Bus 001 Device 004: ID 1b1c:1ba1 Corsair CORSAIR K55 RGB PRO XT Gaming Keyboard

Sganus, 2021-08-15:
The same problem for me...

scott-o-ev, 2021-09-30:
Any word on the update for this issue? I'm having a similar problem with the Corsair CORSAIR K55 RGB PRO Gaming Keyboard.

Thanks.

nikokio, 2021-11-23:
Hello,

Same here, I bought a K55 RGB Pro because K55 is supposed to be supported here but not this model (ToT). I should have check the issues first :p

Bus 001 Device 006: ID 1b1c:1ba4 Corsair CORSAIR K55 RGB PRO Gaming Keyboard

Thanks

vdvb, 2021-11-26:
idem ditto
also bought a K55 RGB Pro , thought it was supported
Thx in advance

SackyMusa, 2021-12-02:
Hello team, unfortunately it's the same for me. My K55 RGB PRO is not recognized by ckb-next. It's a pity, since I've completely switched to Manjaro Linux, and will stay with it. 
Would be very happy to hear from you !!!

tatokis, 2021-12-04:
You can try what I mentioned here and see if you can get anywhere https://github.com/ckb-next/ckb-next/issues/721#issuecomment-983445570

tatokis, 2021-12-04:
Sorry, the correct comment is this one https://github.com/ckb-next/ckb-next/issues/721#issuecomment-981554481

DIYGuy01, 2021-12-27:
> Sorry, the correct comment is this one [#721 (comment)](https://github.com/ckb-next/ckb-next/issues/721#issuecomment-981554481)

a simple replacement in usb.h of id's doesn't help. After installation keyboard is not working at all.

```
//#define P_K55                0x1b3d
#define P_K55                0x1ba4 //K55 RGB PRO
```

marciprete, 2022-03-07:
same problem here, brand new K55 but no luck.

idc77, 2022-04-15:
What can we do to help?

marciprete, 2022-04-16:
I've changed the id of the P_K55 with the output from lsusb. The keyboard is recognized and the GUI shows the map, but unfortunately nothing happens when I type, it looks like the key signals are not received nor interpreted. 
I'm not a C++ programmer, so I don't know what to do to debug. 
Let me know if I can do something to help

localtux, 2022-04-16:
I also have this keyboard since today, unfortunately the workaround does not work with the usb.h. I would be happy if this keyboard is officially supported.

K-Lorenz, 2022-07-07:
So no fix, but if you want to get the annoying watercolor/rainbow to go away you can set the colors in hardware mode.
Holding FN + 1-5 cycles the colors of the 5 zones. 
Holding FN + 6-0 enables the animation color themes.

You can also set makros: https://youtu.be/_gOF_CiBGQo

I know this doesn't really help, but for the time being...

rmoorecpcc, 2022-12-05:
I bought this keyboard a month or so ago and been liking it, at least it does have hardware based color and macro controls but would be nice if ckb-next supported it, That said I did get the source to compile with -DCMAKE_BUILD_TYPE=Debug (had to tinker with the code to add qt6 libraries, and disable the translation files on ubuntu22). Not sure if this will help but there were some bragi errors on the console from launching the daemon:

```
# ./ckb-next-daemon 
ckb-next: Corsair RGB driver 0.5.0-69-gc7d0417
[I] Root controller ready at /dev/input/ckb0
[I] Adding device with vendor = 0x1b1c, product = 0x1ba4, path = /dev/bus/usb/001/005, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6
[I] Connecting CORSAIR K55 RGB PRO Gaming Keyboard at /dev/input/ckb1
[I] Claiming interfaces. name=CORSAIR K55 RGB PRO Gaming Keyboard, firmware=0101, ep_str= 3
[I] ckb1: Claiming 3 interfaces
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/ep_81
[I] Endpoint path /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/ep_81 has wMaxPacketSize 16
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/ep_82
[I] Endpoint path /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/ep_82 has wMaxPacketSize 64
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.2/ep_83
[I] Endpoint path /sys/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.2/ep_83 has wMaxPacketSize 64
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x96. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x2. Device said it's not supported.
[I] ckb1: Pairing id: 00 00 00 00 00 00 00 00 
[I] Setup finished for /dev/input/ckb1
```

JanNowak94, 2023-03-03:
Also can someone mark support correctly on supported devices?  I just did buy one thinking that it was supported by ckb-next (I was wise enough for checking compatibility but not wise enough to check issues).

Also this is keyboard default is mess - G1-G6 return 1-6 by default... which means you cannot remap them by hwdb and no you cannot do that under windows - iCUE does not seem to have way to save binds into hardware memory for this device.

tatokis, 2023-03-03:
Use a `-git` package.

tatokis, 2023-03-29:
For anyone encountering issues with the lighting on this device, please check https://github.com/ckb-next/ckb-next/issues/913#issuecomment-1487872185

closed 2023-03-29 by tatokis