ckb-next/ckb-next #721: Request to add K65 mini

jasonwitty, 2021-06-05:
Request to add K65 mini. please let me know if there is anything I can do to help. report is attached.
[ckb-next-report-1622866167.gz](https://github.com/ckb-next/ckb-next/files/6601997/ckb-next-report-1622866167.gz)

jasonwitty, 2021-06-05:
Bus 001 Device 010: ID 1b1c:1baf Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1baf 
  bcdDevice            0.0c
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR K65 RGB MINI 60% Mechanical Gaming Keyboard
  iSerial                 3 F50019005F74B0BFABA86D261000B034
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     154
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
   can't get debug descriptor: Resource temporarily unavailable
     bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      29
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)

Supercip971, 2021-08-19:
I can also help if you need support for the k65 rgb mini.

kontroll, 2021-11-29:
Is there anything us mere mortals can do to help this along? :)

tatokis, 2021-11-29:
You can try replacing the USB IDs in src/daemon/usb.h for either the K95 Platinum XT or the K100 (in the k100 branch) and see if any of them do anything (with --enable-experimental).
Be prepared for keyboard input to stop working if something isn't right, so it's best not to install ckb-next-daemon while testing, and also have a secondary keyboard plugged in.

JOT85, 2021-11-30:
@tatokis does the software do anything that would be likely to brick the device if something wasn't quite as expected?

tatokis, 2021-12-01:
Typically, no. I haven't had anything like that happen before, but even on the remote chance it'd happen, these devices are usually pretty easily recoverable (unless they're wireless). That said, I do not guarantee anything, and the usual GPL disclaimer applies.

JOT85, 2021-12-01:
Cool, sounds good! How do you go about recovering them? Factory reset or using software?

tatokis, 2021-12-04:
If it's behaving weird, holding down esc while plugging it in should factory reset it.
If the firmware gets corrupt, it either goes into a "safe mode" where you can normally flash new firmware, or it shows up as a usb flash drive where you can, again, copy the firmware over.

You can manually trigger the latter mode by holding down the reset switch that's usually hidden under one of the legs on the keyboard while plugging it in, and then back up the firmware for later use.
If you don't have the reset hole anywhere, then this might not apply. I haven't messed with firmware on devices that don't have it.

But again, it's very likely that the daemon will temporarily break input if it's not set correctly, so just having another keyboard and stopping the daemon (either through systemctl if installed system wide, or killall if ran locally) and a replug should bring it back.

GinjNTonic, 2021-12-12:
> You can try replacing the USB IDs in src/daemon/usb.h for either the K95 Platinum XT or the K100 (in the k100 branch) and see if any of them do anything (with --enable-experimental). Be prepared for [...]

Pardon my noobness, I'm trying to get the K55 Pro XT detected by following this advice.  What do you mean by  "--enable-experimetnal"?

DIYGuy01, 2021-12-27:
a simple replacement in usb.h of id's doesn't help. After installation keyboard is not working at all.

```
//#define P_K55                0x1b3d
#define P_K55                0x1ba4 //K55 RGB PRO
```

antonio-guillen, 2022-01-06:
Same problem here, K55 Pro XT is not detected by version 0.4.4
[image]

pinokaille, 2022-03-22:
> Same problem here, K55 Pro XT is not detected by version 0.4.4 [image]

same :/

Alyx-LeGuen, 2022-09-09:
> You can try replacing the USB IDs in src/daemon/usb.h for either the K95 Platinum XT or the K100 (in the k100 branch) and see if any of them do anything (with --enable-experimental). Be prepared for [...]

I tested this just now and on the k95 Platinum XT, I 've got an error with the keyboard disconnecting in a loop
```
# ckb-next-daemon log
[I] Connecting CORSAIR K65 RGB MINI 60% Mechanical Gaming Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/ep_81
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/ep_82
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/ep_83
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:109): ckb1: Couldn't get bragi device mode. Aborting
[I] Attempting reset...
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:109): ckb1: Couldn't get bragi device mode. Aborting
... loop with the same message
```
No logs from ckb-next app

And with k100-v2 branch, the keyboard works but it is not possible to modify the keys and the leds are off with the following logs:
```
# ckb-next log
Starting "/.../ckb-next/build/bin/gradient"
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
```

```
# ckb-next-daemon log
[I] Connecting CORSAIR K65 RGB MINI 60% Mechanical Gaming Keyboard at /dev/input/ckb1
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.0/ep_81
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.1/ep_82
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-3/3-3:1.2/ep_83
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x15. Device said it's not supported.
[W] bragi_get_property (bragi_common.c:13): Failed to get property 0x16. Device said it's not supported.
[I] ckb1: Pairing id: 00 00 00 00 00 00 00 00 
[I] Setup finished for /dev/input/ckb1
[E] bragi_open_handle (bragi_common.c:193): Failed to open handle 0x0. Error was 0x1
```

Hoping it can help :
└─$ lsusb |grep Corsair
Bus 003 Device 027: ID 1b1c:1baf Corsair CORSAIR K65 RGB MINI 60% Mechanical Gaming Keyboard

tatokis, 2022-09-09:
@Alyx-LeGuen Thank you for testing. Can you re-check that your k100-v2 branch is up to date?
Make sure it contains this commit https://github.com/ckb-next/ckb-next/commit/cc5ee4704f8cb669c1058c79d95a30150591f8e4 .

It looks like the K100 and the K65 RGB mini are extremely similar, and some of the K100s fail setting up the lights in the same way without the changes in that commit.

Alyx-LeGuen, 2022-09-13:
@tatokis Thanks for the quick response, I do have the latest version of branch k100-v2 but unfortunately the lights are still off.

And also, I can modify the bindings it works but I have to redo all the mapping of the secondary keys (FN key).

tatokis, 2022-09-13:
@Alyx-LeGuen What I'm confused about is that in both cases it says "Bragi light init returned not supported". The second time it should say "Bragi alt light init returned error..." instead.

Can you build with `cmake .. -DDEBUG_USB_SEND=1 -DDEBUG_USB_RECV=1` and (making sure the brightness in the GUI is set to 0%) attach the daemon log?
If the brightness is not 0, it'll spam the logs with RGB data.

Alyx-LeGuen, 2022-09-14:
@tatokis
I modified in quickinstall file, line 51 : 
```
cmake -H. -Bbuild -DCMAKE_BUILD_TYPE=Release -DSAFE_INSTALL=ON -DSAFE_UNINSTALL=ON -DDEBUG_USB_SEND=1 -DDEBUG_USB_RECV=1
```
That's it ?

And I generated a report from the application (with 0% brightness) :
[ckb-next-report-1663148797.gz](https://github.com/ckb-next/ckb-next/files/9565093/ckb-next-report-1663148797.gz)

Deamon logs seems very similar to the last one

tatokis, 2022-09-17:
@Alyx-LeGuen Interesting, it's not actually failing. Can you try changing the number 193 to 123 in:
https://github.com/ckb-next/ckb-next/blob/cc5ee4704f8cb669c1058c79d95a30150591f8e4/src/daemon/led_bragi.c#L41

So that it reads:
```c
    LED_CASE_K(P_K100, 123);
```
and then see if the lighting works (make sure the brightness is not set to 0, and that you have undone the changes in quickinstall).

Alyx-LeGuen, 2022-09-19:
@tatokis 
It works ! 😄 🎉 
I've pulled latest version from the k100-v2 branch (d05de233f90656a881db1114f465b2df084770ee) and modified led_bragi.c (193 to 123).

I still have this logs from daemon : 
```
[E] bragi_open_handle (bragi_common.c:193): Failed to open handle 0x0. Error was 0x1
[W] setactive_bragi (device_bragi.c:95): ckb1: Bragi light init returned not supported
```

tatokis, 2022-09-24:
@Alyx-LeGuen Perfect! Those errors are expected.

There are a few things I need to fix first, and then I'll merge support for the K100 and add support for the K65 Mini.

Alyx-LeGuen, 2022-09-24:
@tatokis Cool, let me know if you need to do more tests.
Thank you very much and have a nice day!

tatokis, 2022-10-04:
@Alyx-LeGuen please give the latest code from `k100-v2` a try. You should not need to modify anything.

Alyx-LeGuen, 2022-10-04:
@tatokis Tested just now with commit e0b1ff19b1d6e7021c32ffc17612264fe7e944e8 and the keyboard restarts in a loop

From the daemon log
```
ckb-next: Corsair RGB driver 0.5.0-66-ge0b1ff1
[I] Root controller ready at /dev/input/ckb0
[I] Connecting CORSAIR K65 RGB MINI 60% Mechanical Gaming Keyboard at /dev/input/ckb1
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:154): ckb1: Couldn't get bragi device mode. Aborting
[I] Attempting reset...
[I] Starting input thread for /dev/input/ckb1
[I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/ep_81
[I] Found EP 0x82 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.1/ep_82
[I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.2/ep_83
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:154): ckb1: Couldn't get bragi device mode. Aborting
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:154): ckb1: Couldn't get bragi device mode. Aborting
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:154): ckb1: Couldn't get bragi device mode. Aborting
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:154): ckb1: Couldn't get bragi device mode. Aborting
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:154): ckb1: Couldn't get bragi device mode. Aborting
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:154): ckb1: Couldn't get bragi device mode. Aborting
^C
[I] Caught signal 2
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:154): ckb1: Couldn't get bragi device mode. Aborting
[E] usb_tryreset (usb.c:645): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb1
[I] Caught internal signal SIGUSR2
[I] Closing root controller
[I] Stopping indicator thread for ckb1 (0)
[I] Removed device path /dev/input/ckb0
```

tatokis, 2022-10-04:
@Alyx-LeGuen Sorry, forgot to add it in one place. Try again please. I force pushed.

Alyx-LeGuen, 2022-10-04:
@tatokis Everything seems to be working fine, thank you very much.

I just "lost" the functionality of the "function" keys. For example, I can't do "function+J" keys to have an arrow down or do "function+^" to go to the beginning of the line.
Can I configure this in the software?

tatokis, 2022-10-04:
Unfortunately multi key bindings are not supported. If you can give me a list of the expected Fn shortcuts, I can hardcode them in the daemon until such support is added.

Alyx-LeGuen, 2022-10-04:
That would be really cool, thks a lot
I have the azerty version, I hope it's not annoying for you
(The Fn key can be activated on almost any key, as you can see in the picture)
[image]

The keys I used the most :
" Fn+u " = Up arrow
" Fn+h " = Left arrow
" Fn+j " = Down arrow
" Fn+k " = Right arrow
" Fn+^ " = LineStart
" Fn+ù " = LineEnd
" Fn+& " = F1
" Fn+é " = F2
" Fn+" " = F3
" Fn+' " = F4
" Fn+( " = F5
" Fn+- " = F6
" Fn+è " = F7
" Fn+_ " = F8
" Fn+ç " = F9
" Fn+à " = F10
" Fn+) " = F11
" Fn+= " = F12
" Fn+i " = Sound OFF
" Fn+o " = Increase sound
" Fn+l " = Decreases sound

rapha-pereira, 2023-04-08:
Hey guys, any updates on this?

thelabcat, 2023-06-03:
Another plea for support for this device. Not for me, but for someone I'm helping out.

Thanks for all the work so far, BTW!

devincurrie, 2023-11-21:
Please add support for K65 Pro Mini as well? Thanks!

https://www.corsair.com/us/en/p/keyboards/ch-91a401a-na/k65-pro-mini-rgb-65-optical-mechanical-gaming-keyboard-ch-91a401a-na

flukejones, 2023-11-28:
I recently picked up this keyboard, but mine has an ID of `0x1bc3` as opposed to `0x1baf`. I tried adding next to 0x1baf but I get no input and no dev.

```
Nov 29 12:44:19 fedora systemd[1]: Started ckb-next-daemon.service - Corsair Keyboards and Mice Daemon.
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: ckb-next-daemon 0.6.0-33-gaaf48a03afb0
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: [I] Root controller ready at /dev/input/ckb0
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: [I] Connecting CORSAIR K65 RGB MINI 60% Mechanical Gaming Keyboard at /dev/input/ckb1
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: [I] Starting input thread for /dev/input/ckb1
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: [I] Found EP 0x81 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4.2/3-1.4.2.4/3-1.4.2.4:1.0/ep_81
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: [F] os_inputmain (usb_linux.c:164): Unable to read wMaxPacketSize for /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4.2/3-1.4.2.4/3-1.4.2.4:1.1/ep_83, >
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: [I] Found EP 0x83 at /sys/devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.4/3-1.4.2/3-1.4.2.4/3-1.4.2.4:1.2/ep_83
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: [E] getfwversion (firmware.c:48): Bad input header
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: [W] _start_dev (device.c:151): Unable to load firmware version/poll rate
Nov 29 12:44:19 fedora ckb-next-daemon[76393]: [I] Setup finished for /dev/input/ckb1
Nov 29 12:44:21 fedora ckb-next-daemon[76393]: [E] handle_nkro_key_input (keymap.c:852): Invalid length 16 legacy 0
Nov 29 12:44:21 fedora ckb-next-daemon[76393]: [E] handle_nkro_key_input (keymap.c:852): Invalid length 16 legacy 0
Nov 29 12:44:21 fedora ckb-next-daemon[76393]: [E] handle_nkro_key_input (keymap.c:852): Invalid length 16 legacy 0
Nov 29 12:44:21 fedora ckb-next-daemon[76393]: [E] handle_nkro_key_input (keymap.c:852): Invalid length 16 legacy 0
Nov 29 12:44:22 fedora ckb-next-daemon[76393]: [E] handle_nkro_key_input (keymap.c:852): Invalid length 16 legacy 0
Nov 29 12:44:22 fedora ckb-next-daemon[76393]: [E] handle_nkro_key_input (keymap.c:852): Invalid length 16 legacy 0
Nov 29 12:44:22 fedora ckb-next-daemon[76393]: [E] handle_nkro_key_input (keymap.c:852): Invalid length 16 legacy 0
Nov 29 12:44:22 fedora ckb-next-daemon[76393]: [E] handle_nkro_key_input (keymap.c:852): Invalid length 16 legacy 0
Nov 29 12:44:23 fedora ckb-next-daemon[76393]: [I] Caught signal 15
```

MarieTired, 2024-06-06:
Sorry to post here so long after the issue here has been solved but I'm looking for help with a similar issue. I have a K65 mini as well and tried following the same steps as Alyx but it hasn't worked. The keyboard does not show up in the GUI and input stops working. Interestingly, modifying the ID in line 73 in usb.h to the one that shows up when I run lsusb makes it so that the keyboard is detected in the GUI but won't have input. The code in question is 

`#define P_K65_MINI           0x1baf`

Where I changed `0x1baf` to `1bd7`

I know that the K65 mini is technically not supported yet, but I saw it working for someone else and started tinkering. Any help would be greatly appreciated and while I'm here, huge thanks to the devs of ckb-next.

ioletsgo, 2024-07-03:
is there any potential ETA for support for this keyboard?

blazecoding2009, 2024-08-29:
Any updates? Would love to be able to use custom profiles on my linux setup

MrAwesome, 2024-09-13:
Would love this as well! Happy to chip in any coding that needs to happen, just don't know where to start.

I tried pulling the k100-v2 branch and modifying the ID to what my system reports (`1bc3`), but no luck. Nothing shows up in the interface, and no relevant messages show up in the logs.

Here are the diagnostics for my system with the keyboard plugged in over USB:

[ckb-next-report-1726261439.txt](https://github.com/user-attachments/files/16999352/ckb-next-report-1726261439.txt)

MrAwesome, 2024-09-14:
Realized that I needed to enable experimental for the K65 code to run. With that and the updated ID set:

* the keyboard behaves erratically
* the logs are full of the bragi errors seen above
* the ckb-next UI still doesn't see the keyboard

[ckb-next-report-1726301103.txt](https://github.com/user-attachments/files/17001729/ckb-next-report-1726301103.txt)

MrAwesome, 2024-09-14:
Actually, oops - that was on master. On `k100-v2`, with experimental and the new ID, the keyboard just has no input at all (and still does not appear in the UI).

gravityfargo, 2025-05-13:
Added in bd76bae26051b871dcd66eac9154b762c0d941f9