ckb-next/ckb-next #722: ckb-next not recognising Corsair K60 keyboard

joonapalaste, 2021-06-11:
I am using an HP Z-420 workstation running Fedora 32 Linux. I have a Corsair K60 keyboard and am trying to use ckb-next version 0.4.4.1 to control it.
However, when ckb-next starts, it says "No devices attached". It seems it fails to recognise my keyboard.
I will try to attach the log files from ckb-next to this issue.

joonapalaste, 2021-06-11:
[ckb-next-report-1623406318.gz](https://github.com/ckb-next/ckb-next/files/6637749/ckb-next-report-1623406318.gz)

joonapalaste, 2021-06-11:
ckb-next-dev-detect - Fri Jun 11 13:11:38 EEST 2021

```
Kernel: Linux teletran-1 5.11.11-100.fc32.x86_64 #1 SMP Tue Mar 30 16:53:59 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

OS Version:

USB Devices:
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable

Bus 002 Device 003: ID 1b1c:1bad Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1bad 
  bcdDevice            4.13
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard
  iSerial                 3                                             
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
Device Status:     0x0000
  (Bus Powered)

Dmesg on current boot:
Jun 10 17:07:38.302024 teletran-1 kernel: usb 2-1.1: New USB device found, idVendor=1b1c, idProduct=1bad, bcdDevice= 4.13
Jun 10 17:07:38.302419 teletran-1 kernel: usb 2-1.1: Product: CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard
Jun 10 17:07:38.302574 teletran-1 kernel: usb 2-1.1: Manufacturer: Corsair
Jun 10 17:07:38.302864 teletran-1 kernel: input: Corsair CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1B1C:1BAD.0001/input/input5
Jun 10 17:07:38.355018 teletran-1 kernel: hid-generic 0003:1B1C:1BAD.0001: input,hidraw0: USB HID v1.11 Keyboard [Corsair CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard] on usb-0000:00:1d.0-1.1/input0

Dmesg on last boot:
Apr 02 21:17:59.308031 teletran-1 kernel: usb 2-1.1: New USB device found, idVendor=1b1c, idProduct=1bad, bcdDevice= 4.13
Apr 02 21:17:59.308398 teletran-1 kernel: usb 2-1.1: Product: CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard
Apr 02 21:17:59.308546 teletran-1 kernel: usb 2-1.1: Manufacturer: Corsair
Apr 02 21:17:59.308843 teletran-1 kernel: input: Corsair CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.1/2-1.1:1.0/0003:1B1C:1BAD.0001/input/input5
Apr 02 21:17:59.361036 teletran-1 kernel: hid-generic 0003:1B1C:1BAD.0001: input,hidraw0: USB HID v1.11 Keyboard [Corsair CORSAIR K60 RGB PRO Low Profile Mechanical Gaming Keyboard] on usb-0000:00:1d.0-1.1/input0

Daemon on current boot:
-- Logs begin at Sun 2020-12-20 15:53:41 EET, end at Fri 2021-06-11 13:11:38 EEST. --
Jun 10 14:07:55.235646 teletran-1 systemd[1]: Started Corsair Keyboards and Mice Daemon.
Jun 10 14:07:55.254298 teletran-1 ckb-next-daemon[853]: ckb-next: Corsair RGB driver 0.4.4-1-g50ee67d
Jun 10 14:07:55.254536 teletran-1 ckb-next-daemon[853]: [I] Root controller ready at /dev/input/ckb0

Daemon on last boot:
-- Logs begin at Sun 2020-12-20 15:53:41 EET, end at Fri 2021-06-11 13:11:38 EEST. --
Apr 02 18:18:14.951262 teletran-1 systemd[1]: Started Corsair Keyboards and Mice Daemon.
Apr 02 18:18:14.982723 teletran-1 ckb-next-daemon[852]: ckb-next: Corsair RGB driver 0.4.4-1-g50ee67d
Apr 02 18:18:14.983106 teletran-1 ckb-next-daemon[852]: [I] Root controller ready at /dev/input/ckb0
Apr 02 19:34:15.479798 teletran-1 ckb-next-daemon[852]: [I] Caught signal 15
Apr 02 19:34:15.479798 teletran-1 ckb-next-daemon[852]: [I] Closing root controller
Apr 02 19:34:15.479798 teletran-1 ckb-next-daemon[852]: [I] Removed device path /dev/input/ckb0
Apr 02 19:34:15.402104 teletran-1 systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
Apr 02 19:34:15.431104 teletran-1 systemd[1]: ckb-next-daemon.service: Succeeded.
Apr 02 19:34:15.431333 teletran-1 systemd[1]: Stopped Corsair Keyboards and Mice Daemon.

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/pid
853
/dev/input/ckb0/version
0.4.4-1-g50ee67d
/dev/input/ckb0/connected

Environment:

Kernel command line:
BOOT_IMAGE=(hd5,gpt2)/vmlinuz-5.11.11-100.fc32.x86_64 root=/dev/mapper/fedora-root ro resume=/dev/mapper/fedora-swap rd.lvm.lv=fedora/root rd.lvm.lv=fedora/swap rhgb quiet

GUI ldd:

```

Ravenslofty, 2021-06-11:
As the [Supported Hardware](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware) page says, we do not presently support the K60.

joonapalaste, 2021-06-11:
Are there any plans to support the K60?

Logan007, 2021-06-13:
> Are there any plans to support the K60?

Not sure about plans but I have seen some other issues already asking for it. #705 #697

Makogan, 2021-09-02:
Just bumped into this I also would like support for the k60. And in the meanwhile, is there a way to make the lights stay at a constant color? It is very distracting.

yaliv, 2021-10-15:
> Just bumped into this I also would like support for the k60. And in the meanwhile, is there a way to make the lights stay at a constant color? It is very distracting.

Just press <kbd>Fn</kbd>+<kbd>0</kbd>
You know that, don't you?

joonapalaste, 2021-12-05:
> > Just bumped into this I also would like support for the k60. And in the meanwhile, is there a way to make the lights stay at a constant color? It is very distracting.
> 
> Just press Fn+0 You know that, don't you?

I just tried to press Fn+0 to stop the blinking. Nothing happened. The keyboard just keeps on blinking.

tatokis, 2021-12-05:
Either your keyboard is in BIOS mode and you need to take it out of it, or the firmware on it is corrupt. That's why it's blinking.
Plug it in on a computer with iCUE.

MatthiasSeu, 2022-03-08:
Hello!
Is there any new news about that! I have successfully switched from Windows to Ubuntu, but that I still miss.

kind regards
Matthias

yaliv, 2022-03-14:
@MatthiasSeu it works now:
https://github.com/ckb-next/ckb-next/issues/697#issuecomment-1066095519

And I've confirm it myself :wink::thumbsup:

joonapalaste, 2022-03-15:
> Either your keyboard is in BIOS mode and you need to take it out of it, or the firmware on it is corrupt. That's why it's blinking. Plug it in on a computer with iCUE.

I can't seem to be able to take the keyboard out of BIOS mode. The keyboard does not have a Windows lock key. I have tried unplugging the keyboard and replugging it, either holding the S and B keys, or holding the Esc key. Still, when I replug the keyboard, the four indicator lights flash a few times and the keyboard keeps flashing colours. The keyboard itself works as such, I am typing on it right now.

joonapalaste, 2022-07-23:
I am coming back to this issue. I downloaded the latest updated version of ckb-next 0.5.0 and installed it. When I ran it, it didn't recognise my keyboard. But then I unplugged the keyboard and plugged it back in holding the ESC key, and the indicators didn't blink. Now when I ran ckb-next again, it recognised the keyboard. I haven't tried anything else than change lighting modes yet but that seems to work OK.
Either there was a bug in the previous version of ckb-next or I somehow failed to unplug and replug the keyboard correctly.

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo