ckb-next/ckb-next #724: Expose "turn off lights when idle

Atrate, 2021-06-17:
## What is the program's current behaviour?

## What would you like it to do instead?

## Any other notes & comments?

Atrate, 2021-06-17:
Sorry, accidentally pressed enter, will update the OP

ykarrde, 2023-09-10:
This.

Or at least the possibility to change the backlight / dimming value via commandline.

Got a K95 Premium a few days ago, and what I have done with my old keyboard (for years now) is not possible anymore.

For example with my old keyboard I had scripts that every time locking my screens and/or send all to sleep mode not only immediately blanked all screens but also (stored the keyboard backlit value to reapply it afterwards and) turned the Keyboard light off. Basically using brightnessctl:
`brightnessctl --device='rgb:kbd_backlight'
Device 'rgb:kbd_backlight' of class 'leds':
        Current brightness: 127 (50%)
        Max brightness: 255
`

Now, as a crude workaround, I had to make a profile mode named "off" with every key set to literally black and apply that.
`ckb-next -p "K95P" -m "off"` and then reset that back to Quot on Quot "default" when unlock / waking back up.
Unfortunate I can not even get the old/used mode (stored) before switching to off. (Tried `echo "get :name" > /dev/input/ckb1/cmd` and similar commands listed in the WIKI but that did not work. The notify0 file remained at least empty.)