ckb-next/ckb-next #727: Issue after resuming my Laptop (Corsair K70)

edaustin, 2021-06-20:
Issue with key bindings after a successful resume from suspend (these are macros I use with the PyCharm IDE).

But this is an excellent program and saves me much time with the macros! Many thanks!

**### Source
How and where did you get this program, e.g.:**
Built package from source.

**## What is the program's current behaviour?**
Works fine until I resume from a suspend, it then loses all key bindings, but otherwise works fine (everything else works including the volume control etc.)
I restart the program with no success.
However it works fine after a reboot.

**## Any other notes & comments?**
Laptop is a ThinkPad S2 (i5-6200U, USB2 and USB3) - I connect keyboard through a USB2 hub
OS is Ubuntu 20.04.2 LTS with MATE
Keyboard is a K70 Vengeance.

edaustin, 2021-06-22:
Are bindings actually saved to the keyboard? If so, could this be an issue with the keyboard?
Many thanks!

Ravenslofty, 2021-06-22:
No, key bindings are handled by ckb-next.

edaustin, 2021-06-26:
Anybody solved this?
Same problem with Ubuntu, this time with xfce, on a ThinkPad P50.

tatokis, 2021-06-26:
First of all, please do not spam the issue tracker.
Secondly, please generate and attach a report after the issue has occurred, as was requested by the issue template.

jsmtux, 2021-11-19:
I have the same issue on Ubuntu 21.10. Logs are attached:
[ckb-next-report-1637311004.gz](https://github.com/ckb-next/ckb-next/files/7568872/ckb-next-report-1637311004.gz)
These are my running processes with ckb in their name:
`root        1865  0.0  0.0 158940  2428 ?        Ssl  nov13   0:40 /usr/bin/ckb-next-daemon
jsmtux      4500  0.1  0.7 2427676 120636 ?      Sl   nov13  15:20 /usr/bin/ckb-next --background
jsmtux     35732  0.0  0.0   3752  1364 ?        S    nov15   1:55 /usr/lib/x86_64-linux-gnu/ckb-next-animations/life --ckb-run
jsmtux     61015  0.0  0.0   9288  2500 pts/0    S+   09:38   0:00 grep ckb
`

gravityfargo, 2024-05-02:
This issue has been marked as stale due to age and/or lack of activity.  Please be sure to upload a generated report.

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-02 by gravityfargo