ckb-next/ckb-next #729: Add message about events not changing mode when ckb-next is open

Zettymaster, 2021-07-06:
I was trying out the event manager and - to my surprise nothing happened!

That was until I closed (read as minimized to tray) ckb-next. then everything went smoothly, took me a while to figure out.

So my idea to prevent this sort of confusion is to add a simple one liner to the Event Manager Window, saying something along the lines of "While ckb-next is open, events will not change the selected mode" (better suggestions welcome). I'd make a PR, but I want to learn Qt before actually submitting PRs, so you don't have to correct my beginner mistakes.

Creative-user-name, 2021-07-16:
Which keyboard and ckb version are you using? I'm on a k95 Platinum and it works for me while ckb is open...
Maybe something went wrong on your install? 
If this is the case, this issue would be unecessary.
(I don't have any coding experience, that's why I am so unconstructive, I'm just trying to help. And I don't know, wether it  has something to do with the version of the keyboard)

Zettymaster, 2021-07-17:
well. this is awkward... now it works completely fine. i didnt even install anything

closed 2021-07-17 by Zettymaster