ckb-next/ckb-next #730: Add Saving to Hardware for K95 Platinum

Creative-user-name, 2021-07-14:
As for the 4.4. release, Hardware Playback (as saving to Hardware is called to my knowledge) isn't supported for the k95 Platinum.

That's why I'd love to see it added (plus: the windows software is a pain in the butt) and I think many others do as well. 

I'm new to GitHub and have little to no coding/compiling experience, although I'm very comprehensive and in possession of the Keyboard in question.
I'm running Pop!Os 20.04 (which doesn't really differ from Ubuntu) and am on my way to be a poweruser. I already have Wireshark ready and will give any logs/Information if needed.

Ravenslofty, 2021-07-14:
#136.

The problem with something like this is that
a) I do not have access to the hardware to debug errors, and while we rely heavily on volunteers for things like this, they have a nasty habit of disappearing on us.
b) for several years now I've been running Windows, and that makes ckb-next development challenging for hopefully obvious reasons.

Creative-user-name, 2021-07-14:
That really sound nasty, and I hope, I'm the first one to assure you, that I'm not just gonna drop you. I would REALLY like to see this added.
Yes, the reasons are obvious for it to be a pain to develop in windows, which I hope to compensate for by telling you to just hit me with any request.

I do have to warn you though: I will definitly answer you once a  day, but I  sadly have to watch my A-Levels as well, so I can't go hours at a time. So if you are not up for this, I will definitly understand

Ravenslofty, 2021-07-14:
You can find me on IRC (#ckb-next on Libera chat); there's a button at the
top of the readme.

Creative-user-name, 2021-07-14:
I clicked on "switch to libera chat" and am very confused.. How do I contact you there?

Ravenslofty, 2021-07-14:
Yes, that's right.

An alternative if you for some reason don't like IRC is that I'm available on Discord or Telegram.

Creative-user-name, 2021-07-15:
[k95.zip](https://github.com/ckb-next/ckb-next/files/6822646/k95.zip)

Creative-user-name, 2021-07-15:
[k95(1).zip](https://github.com/ckb-next/ckb-next/files/6823426/k95.1.zip)

Creative-user-name, 2021-07-15:
[k95(realone).zip](https://github.com/ckb-next/ckb-next/files/6823444/k95.realone.zip)

Creative-user-name, 2021-07-16:
[k95save_to_hardware.zip](https://github.com/ckb-next/ckb-next/files/6831688/k95save_to_hardware.zip)

Creative-user-name, 2021-07-16:
[k95_static_save.zip](https://github.com/ckb-next/ckb-next/files/6831805/k95_static_save.zip)

Creative-user-name, 2021-07-16:
[k95_white_static.zip](https://github.com/ckb-next/ckb-next/files/6831979/k95_white_static.zip)

Ravenslofty, 2021-07-16:
Thanks to the help of a friend of mine, I now have a reference white static
profile to compare to what ckb-next does. This has let me fix a couple of
bugs, and so I'd appreciate it if you could give it another go.

Creative-user-name, 2021-07-16:
Okay, will do

Creative-user-name, 2021-07-16:
[k95-xattempt-white-static.zip](https://github.com/ckb-next/ckb-next/files/6833149/k95-xattempt-white-static.zip)

Creative-user-name, 2021-10-18:
Closing the issue due to inactivity

gravityfargo, 2024-05-02:
Duplicate of #95

closed 2024-05-02 by gravityfargo