ckb-next/ckb-next #732: Opening ckb-next causes Cinnamon to crash

zfr77, 2021-07-16:
## General information

Using
Linux Mint 19.3 Tricia
Cinnamon 4.4.8

### Source
How and where did you get this program, e.g.:
ppa:tatokis/ckb-next

## Logs & crash reports
[ckb-next-report-1626448882.gz](https://github.com/ckb-next/ckb-next/files/6831720/ckb-next-report-1626448882.gz)
[cinnamon-crash-report.txt](https://github.com/ckb-next/ckb-next/files/6831722/cinnamon-crash-report.txt)

.xsession-errors has the following:
`(nm-applet:1717): Gdk-CRITICAL **: 16:15:05.068: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed

(mintUpdate.py:2177): Gdk-CRITICAL **: 16:15:05.068: gdk_window_thaw_toplevel_updates: assertion 'window->update_and_descendants_freeze_count > 0' failed`

## What is the program's current behaviour?
When I open ckb-next after the PC has been running for some time, Cinnamon crashes and goes to fallback mode. After restarting Cinnamon, I can open ckb-next fine.
It seems to open fine if I try immediately after system restart.

## What is the expected behaviour of the program?
ckb-next opens without Cinnamon crashing.

## Any other notes & comments?

tatokis, 2021-07-16:
You might want to raise this to the Linux Mint developers. Under no circumstance should any application cause your desktop to crash, so I doubt it's our issue, even if ckb-next is doing something "unexpected".

zfr77, 2021-07-16:
I opened an issue with Cinnamon.

zfr77, 2021-07-16:
I managed to get the exact steps to reproduce. I posted it in the Cinnamon issue

(Using VLC Media Player 3.0.8)

- Have ckb-next start at login. Make sure "Delay application start" in its settings is enabled. This way when the OS starts, ckb-next icon should appear in the panel.
- Do NOT open ckb-next. Just make sure the icon is in the panel.
- Open a movie file with VLC Media player (I tried a few different ones, seemed to work in all). Then close it.
- Click on the icon to open ckb-next. Cinnamon crashes.

zfr77, 2021-07-22:
Can't replicate it on latest Cinnamon. I'll close this.

closed 2021-07-22 by zfr77