ckb-next/ckb-next #733: Window ("Event") manager doesn't work proberly

Creative-user-name, 2021-07-18:
## General information

### Source
How and where did you get this program, e.g.:
  * Built manually using cmake .. and make -j8 and then install.sh I believe (I don't quite remember) just 2/3 days ago 

## Logs & crash reports

Attach:
 
[ckb-next-report-1626601739.gz](https://github.com/ckb-next/ckb-next/files/6835979/ckb-next-report-1626601739.gz)

## What is the program's current behaviour?

After a reboot (and sometimes inmidst a session), the window manager doesn't change my profile upon switching to a different window. The report right now was made after a reboot (the other things haven't occured again yet) and the issue here was solved (as usual) by closing and reopening ckb-next.

This also sometimes happens inmidst a session and is resolved by either

- restarting ckb-next
- moving the window in question around

## What is the expected behaviour of the program?

To switch the profile when going into a different window

## Any other notes & comments?

A variation of this problem is, that the window manager changes to the profile it was supposed  to switch to before changing to a new window. In other words

New window -> nothing happens (supposed to witch to profile A) -> another new window -> changes to profile A

Linux Version: PopOs, happens on Ubuntu 20.04 LTS as well. Kernel: 5.11.0-7620-generic

Creative-user-name, 2021-07-18:
[ckb-next-report-1626618098.gz](https://github.com/ckb-next/ckb-next/files/6836414/ckb-next-report-1626618098.gz)
Now it stopped inmidst session so that I had to restart ckb twice

SkyKwas, 2022-05-04:
I have the same issue.
The mode event manager does not work on first boot. After a restart of the program, it works as intended.

I installed ckb-next (version 0.4.4) from the terminal after installing its PPA.

SkyKwas, 2022-06-08:
Update:
This issue seems to be fixed in the v0.5.0 release.

In the "Important bugfixes" list, one of the fixed bugs is described:
"Window detection is more reliable and works correctly on system boot"

After updating to v0.5.0, ckb-next's event manager seems to work on boot without having to close and open it. It does in fact seem to be more reliable too--I was able to Alt-Tab between windows with different events and ckb-next applied the correct event to each window without bugging-out.

I should mention that I didn't set up a test specifically designed to test the reliability of ckb-next, I'm just talking about my daily use of ckb-next. I haven't had any hiccups yet.

closed 2022-06-12 by tatokis