ckb-next/ckb-next #734: Issue with Ubuntu 21,..04 and Corsair K70

Bass850, 2021-07-20:
First of all thanks for your great software,. I have been using it for a while to full satisfaction.
However, after upgrading to Ubuntu 21.04 all of a sudden my keyboard will not light up anymore after starting CKB-next. 
This used to work perfectly before (even with profiles i created by myself). 
When I start PC keyboard will display wave lights, but as soon as CKB-next starts it turns of to no lights. Wheb i quit CKB or uninstall it the keyboard goes back to rainbow wave.
The app window does work and i can select profiles etc.. I t also shows keyboard device in settings window. Just nothing happens when i select profile ()lights stay off).
I tried checking logs like syslog but cannot find anything pertinent except maybe:
ckb-next.autostart.desktop[3472]: qt.gui.icc: Unsupported ICC profile class 70727472
ckb-next.autostart.desktop[3472]: QPngHandler: Failed to parse ICC profile
I also tried reinstalling multiple times from master github ppa. but to no avail.
Any clue what I could do to get this working again?

Bass850, 2021-07-20:
Sorry i'm an idiot. Just googled one more time and found post about somebody pressing brightness key on keyboard. I pressed it and hey presto: effects are back :-)

Confusing though that this only has effect when CKB is active.

Hopefully htis helps some other schmo like me in future ;-)

closed 2021-07-20 by Bass850

tatokis, 2021-07-20:
What version of ckb-next were you running when this happened?

Bass850, 2021-07-20:
I was running version that is in standard repos of ubuntu

tatokis, 2021-07-20:
Alright, thanks. That version is broken, but I doubt anyone will fix it for 21.04.