ckb-next/ckb-next #735: Corsair Corsair VOID PRO Wireless Gaming Headset

User10987654321, 2021-07-24:
# Before you proceed

- Check if your device is listed in the [wiki](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware).
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

---

Enter the name of your device here and attach the report generated above.

Kernel: Linux Kali 5.10.0-kali9-amd64 #1 SMP Debian 5.10.46-1kali1 (2021-06-25) x86_64 GNU/Linux

OS Version:
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        2021.2
Codename:       kali-rolling

USB Devices:

Bus 001 Device 004: ID 1b1c:0a14 Corsair Corsair VOID PRO Wireless Gaming Headset
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0a14 
  bcdDevice            0.00
  iManufacturer           3 Corsair
  iProduct                4 Corsair VOID PRO Wireless Gaming Headset
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x010d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 Corsair VOID Wireless Gaming Dongle
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0047
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6

mattmattox, 2022-01-03:
I second this

droide02, 2022-05-23:
I have the same headphone and I'm waiting for the same thing.

Martan404, 2023-12-04:
[ckb-next-report-1701676656.gz](https://github.com/ckb-next/ckb-next/files/13543437/ckb-next-report-1701676656.gz)
Here's my report in case it helps

Warrioor, 2023-12-28:
I also need this

Martan404, 2024-05-12:
> [ckb-next-report-1701676656.gz](https://github.com/ckb-next/ckb-next/files/13543437/ckb-next-report-1701676656.gz) Here's my report in case it helps

I realized that I forgot to attach my headset to my computer with USB. Here is my full report
[ckb-next-report-1715529666.gz](https://github.com/ckb-next/ckb-next/files/15286910/ckb-next-report-1715529666.gz)