ckb-next/ckb-next #736: Enabling ckb-next-daemon disables corsair rgb pro mouse.

SilicaAnimusNormanni, 2021-07-29:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If [possible,] try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

## General information
OS is Garuda Linux, fresh install of less than 24 hours.
If I do not use ckb-next, my mouse works, but only as a regular mouse.
If I install ckb-next and start the daemon, my mouse stops responding. Lights turn off. 
Using xinput test, I can clearly see, as soon as the daemon starts, its not getting anything from the mouse.
If I launch it without the daemon, i can get ckb-next to launch, but obviously it tells me it needs the daemon.
If i have the daemon running and launch it, i get qPngHandler errors in the terminal, and it appears that ckb-next sees the device for a second, then does not, then does, and repeats this over and over.
Managed to get a report both with and without daemon running.
I do not get mouse back till I have stopped the daemon, and unplugged/replugged in the mouse.
I have tried the usbcore.quirks bit, no change there. 

### Source
How and where did you get this program, e.g.:
I have tried both package from aur and manually building the latest master.

## Logs & crash reports
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather
[ckb-next-report-1627521242.gz](https://github.com/ckb-next/ckb-next/files/6897265/ckb-next-report-1627521242.gz) Without Daemon
[ckb-next-report-1627521614.gz](https://github.com/ckb-next/ckb-next/files/6897283/ckb-next-report-1627521614.gz)
With daemon running
## What is the program's current behaviour?
T
## What is the expected behaviour of the program?
to at least not stop the mouse from working on launch. I do briefly see it show the mouse shape/buttons, but then it disappears again.
## Any other notes & comments?
I have gone on my laptop and used iCue to make sure it has the most recent firmware (and actually removed the hardware profiles in case they were corrupted)

tatokis, 2021-07-30:
For some reason the connection between the mouse and the laptop isn't good enough and the linux kernel is struggling with it.
Try different ports, try plugging in directly without any KVM switches/docks/USB hubs, and possibly try the usbcore quirks from the wiki.

>Jul 28 14:00:09.882804 Othinn ckb-next-daemon[3223]: [W] os_usbrecv (via firmware.c:24): ckb1: Timeout while waiting for response

This does not look like a ckb bug.

SilicaAnimusNormanni, 2021-07-30:
THANK YOU.
had to do some USB swapping (I've got a few dead ports) but plugging directly in instead of the usb hub did successfully work.

closed 2021-07-30 by SilicaAnimusNormanni