ckb-next/ckb-next #741: Support Need for IronClaw Mouse - No Devices connected

Jeff013, 2021-08-29:
Hey Guys,

Having some issues with the IronClaw mouse.  Hit a road block. I tried the Troubleshooting section but could not resolve the issue.

Tired both options. usbcore and usbhid.quirks.

GRUB_TIMEOUT=2
GRUB_DISTRIBUTOR="Arch"
GRUB_CMDLINE_LINUX_DEFAULT="loglevel=3 usbcore.quirks=1b1c:1b4c:gn"
GRUB_CMDLINE_LINUX=""

Running lsusb
Bus 001 Device 007: ID 1b1c:1b4c Corsair CORSAIR IRONCLAW RGB WIRELESS Gaming Mouse

[deviceinfo.txt](https://github.com/ckb-next/ckb-next/files/7072790/deviceinfo.txt)

[ckb-next-report-1630251320.gz](https://github.com/ckb-next/ckb-next/files/7072789/ckb-next-report-1630251320.gz)

superdregon, 2021-09-02:
You have to start the ckb-next-daemon with experimental features. I really don't relember the command, but I think it is sudo ckb-next-daemon --enable-experimental.

Jeff013, 2021-09-02:
Thanks that worked.

closed 2021-09-02 by Jeff013