ckb-next/ckb-next #742: Corsair Ironclaw RGB WIreless not recognized even though it is supported.

superdregon, 2021-08-30:
## General information
I have a Corsair Ironclaw RGB WIreless mouse, and it is suppostly supoorted, but for some reason it isn't being recognized.

### Source
How and where did you get this program, e.g.:
  * Built manually using quickinstall script

## Logs & crash reports
[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/7075899/ckb-next-dev-detect-report.gz)

## What is the program's current behaviour?
Doesn't recognize supported mouse.

## What is the expected behaviour of the program?
Recognize supported mouse.

## Any other notes & comments?
Tried to make it recognized by connecting via wired and via wireless mode, none worked.

superdregon, 2021-09-01:
Ok found fix already. Just start ckb-next-daemon with experimental features.

closed 2021-09-01 by superdregon

NicTanghe, 2023-01-27:
Hello how do i start it with experimental features ?

Alethephobe, 2024-03-08:
I'm experiencing the same issue.
Corsair Ironclaw RGB WIreless mouse that isn't being recognized. I just see the lines "No devices connected"

I tried editing the ckb-next-daemon.service and adding: ExecStart=/usr/bin/ckb-next-daemon **--experimental** 
Reloaded and restarted the daemon.

Is there anything else I must do?

Alethephobe, 2024-03-08:
@NicTanghe  https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#enabling-support-for-experimental-devices