ckb-next/ckb-next #744: Event manager doesn't work

ClemGlacee, 2021-09-12:
## General information

CKB does not switch profiles on autostart and has to be close (quit) and reopened to work and/or the user have to manually click on another profile for it to switch automatically based on events 

### Source
How and where did you get this program, e.g.:
  * Package from AUR

## Logs & crash reports

[ckb-next-report-1631441438.gz](https://github.com/ckb-next/ckb-next/files/7149394/ckb-next-report-1631441438.gz)

## What is the program's current behaviour?
CKB won't switch profiles depending on the program on the first run (autostart)

## What is the expected behaviour of the program?
Changing profile depending on the program setup in "manage events"

## Any other notes & comments?
Why can't it load properly the first time and switch profiles automatically without having to quit the program and reopen it

ClemGlacee, 2021-09-12:
#702 was the same issue but it never got fixed apparently

ClemGlacee, 2021-09-12:
I have no issue or errors when closing ckb-next (checked in the terminal)
```Disconnected from PulseAudio 6
Disconnecting from X
XCB EWMH Thread Quitting...
```

Although this is being spammed in the terminal when ckb is opened
```qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile
qt.gui.icc: Unsupported ICC profile class 70727472
QPngHandler: Failed to parse ICC profile```

SteffanoP, 2021-09-28:
Same Issue here, even if I delay the application start...

jdpnielsen, 2021-11-02:
I'm running Gnome 3.38.4-1ubuntu3~21.04.1.

Output is:
```
$ echo $XDG_SESSION_TYPE
echo $XDG_SESSION_DESKTOP
x11
ubuntu
```

ClemGlacee, 2021-11-09:
OS: EndeavourOS Linux x86_64
Kernel: 5.14.16-arch1-1
DE: Plasma 5.23.2‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

zeparu, 2022-03-31:
running into this issue too right now, at first i thought the automatic profile switching didnt work at all, but manually restarting ckb-next after it was auto-started, makes it work

tatokis, 2022-04-01:
Sorry, I really haven't had any time to look into this. Is there a way for someone to get a log from the gui during a start when it's broken? (For example from /var/log/syslog on ubuntu.)

zeparu, 2022-04-02:
this here is a full journalctl log from starting it with the auto-start, to creating a new color mode and setting up the dynamic switching rule https://pastebin.com/eBFUNcAi

tatokis, 2022-04-02:
Should be fixed in the latest code from master (https://github.com/ckb-next/ckb-next/commit/27d0547d09e481ade933bfb3292498bbbe2e7087)

zeparu, 2022-04-02:
i can install `ckb-next-git` and see tomorrow if itll work properly

zeparu, 2022-04-03:
can confirm, that its working properly with switching now, when ckb-next was started with the auto start feature

tatokis, 2022-04-03:
I ended up rewriting some of the window detection code. Can you please test again?

> when is the next release scheduled?

It's not. There are a lot of things that still need to be taken care of.

tatokis, 2022-04-04:
Thank you for testing!

closed 2022-04-04 by tatokis

maplepy, 2022-04-05:
Do you have an ETA so I can download that program back?