ckb-next/ckb-next #745: Corsair K70 RGB TLK

HuTaeh, 2021-09-14:
Hi there,

The keyboard K70 RGB TLK are not supported by ckb-next. No device is detected.
It's possible for you to implement it ?

Thank you for your work
---

* Version ckb-next : 0.4.4-99-g761ba48
* Report ckb-next : [ckb-next-report-1631657442.gz](https://github.com/ckb-next/ckb-next/files/7165434/ckb-next-report-1631657442.gz)

--- 
* Keyboard Name : Corsair K70 RGB TLK Mechanical Gaming Keyboard
* Keyboard Model : RGP0071
* Official Keyboard Web page : https://www.corsair.com/us/en/Categories/Products/Gaming-Keyboards/RGB-Mechanical-Gaming-Keyboards/K70-RGB-TKL-CHAMPION-SERIES-Mechanical-Gaming-Keyboard/p/CH-9119010-NA

BloodyIron, 2021-12-18:
Yeah seriously considering getting this keyboard, and since the last ckb-next release was March... can't really tell when this will get added :/

b0rnwr0ng, 2022-01-10:
I got this keyboard, shows as:
`ID 1b1c:1b73 Corsair CORSAIR K70 RGB TKL CHAMPION SERIES Mechanical Gaming Keyboard`

gtjoseph, 2022-02-01:
I just bought one to replace my broken K65 and was surprised it wasn't supported.  After some tinkering with the source it appears that this keyboard uses a slightly USB interface with a command buffer like the "legacy" devices.

I'll tinker with it more maybe this weekend.

```
CORSAIR K70 RGB TKL CHAMPION SERIES Mechanical Gaming Keyboard
Manufacturer: Corsair
Serial Number: F5001900608D913DAA3518681501903A
Speed: 480Mb/s (high)
USB Version:  2.00
Device Class: 00(>ifc )
Device Subclass: 00
Device Protocol: 00
Maximum Default Endpoint Size: 64
Number of Configurations: 1
Vendor Id: 1b1c
Product Id: 1b73
Revision Number:  1.05

Config Number: 1
	Number of Interfaces: 3
	Attributes: a0
	MaxPower Needed: 500mA

	Interface Number: 0
		Name: usbhid
		Alternate Number: 0
		Class: 03(HID  ) 
		Sub Class: 01
		Protocol: 01
		Number of Endpoints: 1

			Endpoint Address: 81
			Direction: in
			Attribute: 3
			Type: Int.
			Max Packet Size: 16
			Interval: 1ms

	Interface Number: 1
		Name: usbhid
		Alternate Number: 0
		Class: 03(HID  ) 
		Sub Class: 00
		Protocol: 00
		Number of Endpoints: 2

			Endpoint Address: 82
			Direction: in
			Attribute: 3
			Type: Int.
			Max Packet Size: 1024
			Interval: 1ms

			Endpoint Address: 01
			Direction: out
			Attribute: 3
			Type: Int.
			Max Packet Size: 1024
			Interval: 1ms

	Interface Number: 2
		Name: (none)
		Alternate Number: 0
		Class: 03(HID  ) 
		Sub Class: 00
		Protocol: 00
		Number of Endpoints: 1

			Endpoint Address: 83
			Direction: in
			Attribute: 3
			Type: Int.
			Max Packet Size: 64
			Interval: 1ms
```

minneyar, 2022-04-02:
@gtjoseph Out of curiosity, did you have any luck getting this to work?

I also just got this keyboard only to discover it's not supported; I tinkered a little bit with trying to add support for it to ckb-next and discovered that its interface seems to be a little different from other keyboards, and I'm not familiar enough with the driver to know where I should investigate next.

tatokis, 2022-04-04:
@minneyar  Based on the information provided above, it appears to use 1024 byte packets and the same endpoints as a K100. Check out the k100 branch and try adding it in a similar way to the K100 (like this https://github.com/ckb-next/ckb-next/commit/50c3eed16cc3531ef5c7b44c08e40bcf9c13c5ef). Make sure to test on the k100 branch and not master.

minneyar, 2022-04-05:
@tatokis Thanks, that's a big help!  I've managed to get support for the K70 RGB TKL *mostly* working ( https://github.com/minneyar/ckb-next/tree/k70tkl ).  It recognizes all of the keys, I've got an updated key map in the ckb-next GUI, and animations & brightness work, but... the colors are weird.  If I try to set every key to pure white in the GUI, they turn an assortment of yellow and red, with a few black.  If I set them to all red, most of them are red; if I set them to all green, two are green, ten are red, and the rest are black; and if I set them to all green, about half are green and half are black.  I can't really make out any pattern to this other than it seems like there's no blue component to any of them.  Any idea what I should look at to figure out why the colors aren't getting set properly?

tatokis, 2022-04-05:
@minneyar This happens because the led count you set is not correct. If the number is too high, when you set everything to all red, some keys might light up red+green. If the number is too low (all red again), some keys will not light up at all.

Typically if you set the keyboard to all blue and they all light up correctly (blue only), you got the correct number. Even if it looks "right" though, keep testing higher and lower ones to be absolutely sure.

This happens because the data is laid out as RRRR[...]GGGG[...]BBBBB. So if you don't have the correct number of red bytes, the green ones will take their place, or vice versa.

minneyar, 2022-04-05:
@tatokis That makes sense.  It looks like the right value is 193, same as the K100.  All the colors look right now!  The only things that seem suspicious is that the "lock" key never lights up, and the "logo" led seems to just switch between black and white based on whether "lock" is pressed or not.  I noticed that there's some special handling for the colors of the light/lock/mute keys, so I wasn't sure if that's intentional or not.

Otherwise, I think I'm about done with this; should I make a pull request targeting the `k100` branch since that's what I based this on?

PR #809 (2023-06-24): Add support for the K70 RGB TKL
This adds support for the K70 RGB TKL Champion Series keyboard.  Product page is here: https://www.corsair.com/us/en/Categories/Products/Gaming-Keyboards/RGB-Mechanical-Gaming-Keyboards/K70-RGB-TKL-CHAMPION-SERIES-Mechanical-Gaming-Keyboard/p/CH-9119014-NA

Its interface is fairly similar to the K100, but its key layout is closer to the K63.

I've tested this with my own keyboard, and everything seems fully functional.

Fixes #745
closed 2023-06-24 by tatokis