ckb-next/ckb-next #746: [Bug] lighting does not turn off when idle, K95 RGB Platinum (base model, not XT)

VanessaE, 2021-09-15:
Title pretty much says it all.

When the idle timer runs out, the keyboard no longer turns off the lights.  Instead, the animation just freezes in place.

If I then press a key or waggle the mouse, the lighting turns off for a split second, and then the animation resumes.

I am using the default "Wave" animation and Type Lighting, though I do have a few keys fixed at [0,0,0] because I don't use them, so it's not as if the LEDs are all stuck "on" or at one color or anything like that.

The lighting DID turn off when idle when I got the keyboard a month or so ago (it's a refurb from Amazon), so this is a recent change of behavior.

If I turn the brightness down via the keyboard's button or from ckb-next, the various settings all work, but when I select the off setting, the animation freezes, until I step the brightness back up.

The keyboard is 100% functional in all respects that I can see, apart from the above.

I have tried powering-down/rebooting, unplugging the keyboard, deleting the ckb-next config (which indeed reset the keys I intentionally blanked-out), and nothing helps.  I also tried the reset procedure at https://help.corsair.com/hc/en-us/articles/360025465031-How-to-Reset-your-keyboard but that did nothing at all.

Maybe this is a firmware glitch?  ckb-next says it's v3.21 and "up to date", though some googling suggests this isn't the latest version.

## General information

### Source

Debian "Bullseye" standard repositories, ckb-next is v0.4.3

## Logs & crash reports

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/7168790/ckb-next-dev-detect-report.gz)

I see this in the "Daemon on current boot" section of the log (which I generated as root), which may be important:

```
Sep 15 04:20:22.000103 rainbird ckb-next-daemon[569]: [W] os_usbrecv (via firmware.c:24): ckb1: Timeout while waiting for response
Sep 15 04:20:22.000103 rainbird ckb-next-daemon[569]: [W] _start_dev (device.c:76): Unable to load firmware version/poll rate
```

## What is the program's current behaviour?

Completely normal in all respects, except for LEDs not turning off when idle.

## What is the expected behaviour of the program?

Lighting is supposed to turn off when idle.

## Any other notes & comments?

* Yesterday, the keyboard stopped responding unexpectedly, though the animation continued, but unplugging it and plugging it back in sorted it.  The lighting stopped turning off when idle the same day.

* For some reason, the polling rate dropdown disappeared from ckb-next today, pretty sure it was in the Device tab.

tatokis, 2021-09-15:
> Debian "Bullseye" standard repositories, ckb-next is v0.4.3

We do **not** support old versions of ckb-next. There were some bugs related to the automatic turning brightness off on devices, along with all sorts of other issues that have since been fixed. If you want support for older versions, open a bug on the debian bug tracker so that they provide support instead.

If this still happens with the latest code from master, then we can look into it.

> Sep 15 04:20:22.000103 rainbird ckb-next-daemon[569]: [W] _start_dev (device.c:76): Unable to load firmware version/poll rate

It also sounds like you're having general connectivity issues. Try different USB ports, do not use any extensions or USB hubs, and the usbcore quirks in the wiki, although I suspect your kernel might already have them.
Whenever the device doesn't respond back during the initialisation, some features like the polling rate are hidden in the GUI.
Also, 1ms polling rate helps.

VanessaE, 2021-09-15:
Oh, sorry, didn't realize Debian's was that outdated (a little surprising really, as even though my install is tagged as Bullseye, it's actually still the 'testing' branch).

Still, for whatever it's worth:  for some unexplained reason, some of my other USB 3 ports stopped working, turned out to be the kernel's autosuspend feature being a little overzealous.  Edited GRUB config to disable it, rebooted, now the keyboard lighting behaves properly and the poll rate dropdown is back (and of course those other USB 3 ports work now).

closed 2021-09-15 by VanessaE

tatokis, 2021-09-15:
No worries. Glad you got it solved. Would you mind mentioning what parameters you added to the kernel cmdline to fix this? It might help other people too.

VanessaE, 2021-09-15:
I added `usbcore.autosuspend=-1`.

Source: https://askubuntu.com/questions/526082/two-usb-ports-stopped-working

tatokis, 2021-09-15:
Awesome, thanks!

VanessaE, 2021-12-31:
So... for reasons I can't begin to guess, the keyboard lighting started not turning off again.  Actually been that way for a bit, but I'd mostly just been ignoring it since it's not a huge problem.

Today I decided to poke around in the ckb-next window... The "Device" tab indicated that there's new firmware available when I told it to check, and it said that ckb-next v0.4.4 was required in order to use it, thus finally giving me a good enough reason to uninstall the 0.4.3 deb package and build/install ckb-next from source.  I used the instructions linked from the readme, and rebooted when done just to be sure.

Thus, I'm now I'm running up-to-date code.

With that in mind, the keyboard lighting still does not turn off when idle or set to 0% (either from the menu or with the button on the keyboard).

I'm not sure if it matters, but when I try to check for firmware updates now, it throws an error, "There was a problem getting the status for this device" and offers to let me choose a file (but I can't find any official downloads of it).

What do I do now?  Does this issue need to be reopened?