ckb-next/ckb-next #747: Scroll wheel stopped working

idlyby, 2021-09-20:
## General information

### Source
  * AUR package
  * Built manually using quickinstall script

## Logs & crash reports
[ckb-next-report-1632117379.gz](https://github.com/ckb-next/ckb-next/files/7193774/ckb-next-report-1632117379.gz)

## What is the program's current behaviour?
Scrolling with the scroll wheel no longer works upon updating `xf86-input-libinput`.

## Any other notes & comments?
The issue appears to stem from the api changes in `xf86-input-libinput-1.2.0-1`.
- Using xev showed no scroll wheel events when ckb-next-daemon was active. Upon stopping the daemon scroll functionality returned and xev displayed scroll wheel events.
- Watching /dev/input/ckb1/notify1 showed mouse scroll events when the daemon was active.
- Downgrading to xf86-input-libinput-1.1.0-1 restored scroll wheel functionality when the daemon was active.
- There were three commits in particular that seem to have changed scroll wheel: [bf8dc2e2](https://gitlab.freedesktop.org/xorg/driver/xf86-input-libinput/-/commit/bf8dc2e2ed0780b947eccfc9ac3694c518dee605), [ca9042c7](https://gitlab.freedesktop.org/xorg/driver/xf86-input-libinput/-/commit/ca9042c7f08f8f0dc214b9cc19f3243728ec8c4a), [beb94333](https://gitlab.freedesktop.org/xorg/driver/xf86-input-libinput/-/commit/beb94333e1450006942a7f0ee38bc2a2f5719238)

tatokis, 2021-09-20:
Please bring it up with them. All we're doing is sending scroll events through uinput...
https://github.com/ckb-next/ckb-next/blob/761ba48d08b82b07c681ad14ab9c292eb29dec06/src/daemon/input_linux.c#L212-L230

I'll go ahead and pin this because I'm sure we'll get a storm of these issues.

idlyby, 2021-09-20:
No problem, thanks for the prompt reply. Closing the thread.

tatokis, 2021-09-20:
I think it should be kept open until the solution is found.

Although I'd argue that if we're doing something "unexpected", since it worked, it should keep working.
Unfortunately I don't really have time to bisect or look into this at the moment, and won't for a while.

idlyby, 2021-09-20:
Ah apologies I'll leave it open, thanks again!

The issue can be followed here: [gitlab.freedesktop.org](https://gitlab.freedesktop.org/xorg/driver/xf86-input-libinput/-/issues/38)

tatokis, 2021-09-20:
Can you try the latest code from master? This patch should fix it. https://github.com/ckb-next/ckb-next/commit/dc4dc54c5ebac7e4b455d8df35076fc044a581a7

Thanks

idlyby, 2021-09-20:
Thank you for looking into this. Can confirm master is working, both xev and libinput are correctly reporting scroll wheel events.

madipietro, 2021-09-20:
Facing the same problem on ArchLinux, downgrading the xf86-input-libinput package to the mentioned version fixed the problem.

meow6969, 2021-09-23:
noticed that if you rebind the event to something like a keyboard key, it works fine

tatokis, 2021-09-27:
I have already pushed the fix on both ckb-next and ckb-next-git on AUR.
Just rebuild the packages, if you haven't been prompted to update them.

closed 2021-09-27 by tatokis