ckb-next/ckb-next #751: Corsair K60 RGB PRO CH-910D019-NA

lotar1954, 2021-10-01:
The keyboard is not detected by the ckb-next software (latest build included).

`Bus 002 Device 007: ID 1b1c:1ba0 Corsair CORSAIR K60 RGB PRO Mechanical Gaming Keyboard
`

Please let me know if you need further info. In addition to this, if you need a tester for the K60 RGB PRO, count me in. 

[ckb-next-dev-detect-report.gz](https://github.com/ckb-next/ckb-next/files/7270439/ckb-next-dev-detect-report.gz)

Synfoxic, 2021-10-05:
there is already an opened issue for this: https://github.com/ckb-next/ckb-next/issues/720#issue-909961273

Synfoxic, 2021-10-18:
> > there is already an opened issue for this: [#720 (comment)](https://github.com/ckb-next/ckb-next/issues/720#issue-909961273)
> 
> No, that's about K55.
> 
> For K60, there are #697, #705, #722

sorry, my bad

MatthiasSeu, 2022-02-10:
Are there any new news regarding K60 support?

akintelligentmedia, 2022-02-24:
There is some work in progress #796!

akatran, 2022-02-24:
There is some work in progress https://github.com/ckb-next/ckb-next/pull/796!

gravityfargo, 2024-05-02:
Support was added with #798

closed 2024-05-02 by gravityfargo