ckb-next/ckb-next #755: Macro "repeat" function

DweetOnWeed, 2021-10-18:
## What is the program's current behaviour?

right now, you only have a "press" and "release" funtion

## What would you like it to do instead?

a "repeat" function that basically once you press the button, toggles on the macro and when you press the button again, toggles off the macro

## Any other notes & comments?

would love to see this feature implemented in ckb-next! :D

amajauskas, 2022-05-28:
Not only toggle, as iCUE has a "as long as the button is pressed do this" then you can set "this" to be click once every 1ms constantly, which in turn would create a press and hold "rapid click" function. Would be nice to have.

Flakoeh, 2023-09-23:
I love ckb-next, works on my K100 opx (so far) as well. The addition mentioned would really be a great improvement!

CalAlaera, 2024-05-24:
I'm transitioning to Linux from Windows and this is the only feature of iCUE that I actually miss, it'd be awesome if you could add it for easy rapid clicking please!

Ahajha, 2024-08-14:
+1 for this, I always found this super useful in iCUE.

AdamKafei, 2024-11-13:
+1 Really need this.

fleischie, 2024-11-15:
@Flakoeh @CalAlaera @Ahajha @plinio1980 @AdamKafei and everybody else reading this issue: Thank you for your interest, PRs welcome. 🙂

We are very happy to welcome new contributors and gladly walk you through the code in case of inclarities. But this project is thinly populated as it is and we are neither waiting for new feature requests nor upvotes to them for us to spend our limited time and attention to. (Also we cannot compete with a for-profit enterprise such as the one that provides you iCue). Thank you for your understanding.

CalAlaera, 2024-11-15:
> Thank you for your interest, PRs welcome. 🙂

Unfortunately I don't know how to program in C++. I can barely program in Python. Hopefully someone will be able to help out the project though 👍