ckb-next/ckb-next #760: Heatmap Fading Problems

some0nenew, 2021-11-02:
I set up a Heat map to change the color of pressed keys from white to red. (Frames to fad: 50 ; Presses to full: 1 ; start on pressed key) Worked fine for about 3 months now, since Sunday (31.Oct) the color does not fade anymore. All keys once pressed just stay red. 
Closing the Program and reopening it solves the problem, then the heat map works fine again. I tried the "Delay application Start" setting hoping that this might solve the problem, it did not.

I installed the AUR package via the Manajaro Package manager. Removing and Building it again doesn't solve the problem. It remembered the profile I had set up before so i guess there are some config files that didn't get removed, but i cant find them.

Looking at the timing i guess it could be a problem with the switch from summer to winter time (31.Oct at 3am). I also have minor error in my VPN client, the connection timer starts at -1h. So maybe that's a Problem across my machine (why would restarting the program solve it then tho?), but as I'm rather new to Linux overall I don't really know how i could troubleshoot this...

[ckb-next-report-1635848955.gz](https://github.com/ckb-next/ckb-next/files/7459909/ckb-next-report-1635848955.gz)

Ravenslofty, 2021-11-02:
Here's what I suspect is happening: ckb's animation API promises that time does not go backwards, and I *suspect* that the timezone change was a situation where time did in fact go backwards for the animation.

If that happens, then the `delta` parameter passed to `ckb_time` would be negative, which would then be equivalent to saying "fade after an hour" if I'm reading the code right.

Daylight Savings Time is tricky like this.

some0nenew, 2021-11-03:
After some more testing i found out that disabling the setting "Hardware clock in local time zone" fixes the problem. I found the Setting in the "Manjaro Settings manager".

But that's still not a fix for the underlying problem i guess.

closed 2022-03-29 by tatokis