ckb-next/ckb-next #761: Turn off light permanently

enei-pereira-jr, 2021-11-12:
## What is the program's current behaviour?

Whenever PC turn on/off, the lights from my keyboard start to work automatically even with "turn off" from program.

## What would you like it to do instead?

I'd like to have an option that disable RGB functions permanently until I turn on again. Because, in my case, I use ckb next just for macro settings.

## Any other notes & comments?

N/A

shadow7412, 2022-01-30:
You should be able to achieve that by creating a profile where the main colour is set to black. Would that suffice?

tatokis, 2022-03-29:
> Whenever PC turn on/off, the lights from my keyboard start to work automatically even with "turn off" from program.

This sounds like a bug in the firmware which is out of our control, but as a workaround you should be able to do what @shadow7412 mentioned, but it needs to be saved to hardware (if save to hardware isn't available in ckb-next for your device you'll have to use icue).

closed 2022-03-29 by tatokis