ckb-next/ckb-next #762: K95 Platinum RGB - Weird backlight behavior

tippl, 2021-11-14:
## General information

### Source
How and where did you get this program, e.g.:
  * AUR package ckb-next-git (built from latest commit as of today)

## Logs & crash reports

Attach:
[ckb-next-report-1636883757.gz](https://github.com/ckb-next/ckb-next/files/7533509/ckb-next-report-1636883757.gz)

## What is the program's current behaviour?
- On system boot, after login when the ckb-next applications loads, the keyboard backlight turns off. When i press any key, backlight turns on.
- Using the GUI or bound key on the keyboard, changing the brightness works weird. After changing brightness, change is only applied when i press any key. When selecting off for backlight, pressing any key turns it on at 100% (flickers off and straight to on when using bound brightness key)
- The "Type Lighting" effect seems delayed - turns white after the key is released, not pressed.

## What is the expected behaviour of the program?
- On system boot, the keyboard should stay on if the ckb-next settings are set to 100% brightness. The keyboard realizes this after first keypress.
- Brightness control should work properly.
- Type Lighting used to keep the key white until released, and then it slowly animated back to usual color. Now the first press and hold has the key be usual color, and turns white only after releasing the key.

## Any other notes & comments?
In general, the backlight started acting weird. To me, it seems that the state ckb-next thinks keyboard is in is not properly synchronized with the keyboard. This can be especially seen in the brightness issue, where the brightness changes only after another keypress.

tatokis, 2021-11-29:
Can you please export and attach one of the affected profiles?

tippl, 2021-11-29:
Profile attached here (renamed to .zip so that github accepts it): [Demo.zip](https://github.com/ckb-next/ckb-next/files/7619714/Demo.zip)

I also did some more testing, and i think it might not be a 100% issue with this app after all. Downgrading from ckb-next-git to ckb-next (latest release not commit) exhibits the same issue of keyboard going black after login.

This leads me to believe, it might be a combination of this application and possibly some change in kernel drivers or something similar. Currently i'm running kernel 5.15.5-arch1-1.

tatokis, 2021-12-01:
Yeah, unfortunately I don't seem to be able to reproduce the bug. Maybe try the usbcore quirks as shown in the wiki.
Alternatively, you can try an lts kernel for testing.

Maybe also give this a try in the kernel cmdline `usbcore.autosuspend=-1`.

tippl, 2021-12-01:
Sadly, looks like there are no changes in the behavior no matter what i try.

In order:
- Installed the LTS kernel (linux-lts-5.10.83-1) and rebooted to use it.
- Added the `usbcore.autosuspend=-1` kernel parameter
- Added the usbcore quirk following the wiki

Adding the quirk made the keyboard and mouse unresponsive after boot for a few seconds, otherwise all changes had no effect on the issue. Sadly i'm at a loss on where to debug this.

I can force the issue by rebooting the ckb-next-daemon service, but if it still happens on older kernel and older version of ckb-next, i have no idea what it could be caused by.

All that is left to test on my side is to perform a full system rollback using the archlinux archive and update day by day to see what package broke it. However that will have to wait at least till the weekend.

tippl, 2021-12-03:
Well, i am out of ideas on this. I have tried the following:
- Rolled back the entire system from the Archlinux Archive to packages from 2021-08-01
- Uninstalled any recently installed packages

None of these have changed anything. Nor have i recently updated the keyboard firmware.

As you cannot reproduce the issue, and i've tried everything i can think of, i feel like we can close this issue as unsolved, unless either of us comes up with anything else to try.

tatokis, 2021-12-04:
When you downgraded to 0.4.4, have you tried renaming the config folder before running the GUI? Maybe something is wrong in the config...
You can do `mv ~/.config/ckb-next ~/.config/ckb-next-old` while the GUI isn't running, and right after you have downgraded to the `ckb-next` AUR package, but before you've ran the GUI again.

tippl, 2021-12-04:
Behaves exactly the same. Another thing i've noticed is that the type animation is actually the thing that brings the keyboard back to life.

For example, if i set it to be all red, remove the type lighting and then restart the ckb-next service, the keyboard turns off and stays that way. So the animation doing it's thing is forcing the keyboard to turn back on. Without it, it would stay black.

tatokis, 2021-12-04:
Can you try to, instead of using the GUI when you know this issue will occur, instead do `echo -e "active\nrgb ff00ff" > /dev/input/ckb1/cmd` ?
Basically don't let the GUI run and run that command in its place.
If you have multiple devices handled by ckb it might not be ckb1.

This will tell us if it's an issue with the GUI, or if it's with the daemon or anything at a lower level.

tippl, 2021-12-04:
Looks like it is not an issue with the GUI itself.

Play-by-play of what happens (gona type only the color part of the command)
1. Keyboard is red
2. Run command ff00ff
3. Keyboard is still red
4. Run command ff0000
5. Keyboard turns purple
6. Run command ffffff
7. Keyboard turns red
8. Run command 000000
9. Keyboard turns white
10. Run command ff0000
11. Keyboard turns black

So there seems to be a delay between what you push into the daemon and what the keyboard changes to. The current change does not get input, only the once the new change happens, does the keyboard change to the previous color.

note: performed on the ckb-next-git (reinstalled it since it behaved the same) with gui turned off and service freshly restarted

tatokis, 2021-12-04:
Right, you should only be running the active command the first time, and then only rgb XXXXXX afterwards. Make sure you have a newline at the end of each command, otherwise the daemon won't process it, but I don't think that's the issue here (echo sends a newline at the end by default).

And yes, thank you for doing the testing with the -git package at this point. It will make fixing things easier, as there have been a lot of changes since the last release. Please keep testing with the latest git code from now on, now that we know it doesn't make a difference.

Try compiling ckb-next with -DDEBUG_USB_SEND=1 passed to cmake. It will dump all the packets it sends to the console, so you probably don't want to run the daemon over systemd. Instead, run it through a terminal and look at the output in realtime. It sends quite a few packets for one frame, however, you should be able to tell the difference between, say, 990000 and ff0000.

If when you send the rgb ff0000 command, the daemon in the console spits out packets with multiple 99 bytes in them, instead of ff, then it's definitely a bug in the daemon.
If I recall correctly, the first few packets are red, then the next are green, then blue. I forgot how these old devices send rgb packets.

tippl, 2021-12-04:
Hmm, looks to me like the issue is not with the daemon then.
Sending out 990000 dumps a bunch of packets with 99 bytes
Sending out ff0000 dumps a bunch of ff

Guessing it is between the daemon and the keyboard, or some issue with the keyboard firmware then?

tatokis, 2021-12-04:
Yeah, this seems like a possible kernel issue. Is the keyboard plugged in through a usb hub? Try changing the ports.
Maybe also try resetting the keyboard by unplugging it, then holding down Esc, and plugging it back in while Esc is held down.

tippl, 2021-12-04:
There we go. I feel stupid, but i've tried the thing i should have weeks ago. Tested it on Windows with the official software.
It behaved exactly the same, setting pushed to keyboard did not apply until next option was applied.

So this suggests a borked firmware, ended up having to factory reset it, and then updated it from 3.21 to 3.29 (couldn't even update without the reset). After the reset, it works as it should.

Thanks for the help in troubleshooting this, luckily none of the issues were caused by this software.

closed 2021-12-04 by tippl

tatokis, 2021-12-04:
I didn't realise there was a new firmware out. Can you confirm it works well with ckb-next so that I can update the firmware table?

tippl, 2021-12-04:
Not sure if I can confirm everything, but from some basic testing (animations, bindings, per key lighting) everything seems to work fine.

tatokis, 2021-12-04:
Awesome, thanks! it usually either breaks entirely, or just continues working, so that's sufficient.

I'm glad you got it fixed, and now for the future, we know what to suggest to people that might have a similar issue.
I'll update the table, and hopefully this bug isn't present in the newer firmware, so when people who use ckb exclusively update, they won't encounter this.

tippl, 2021-12-04:
Don't think it was really a bug in the firmware itself, i feel like something in the memory had to have corrupted. Mainly because the firmware update couldn't even apply until i performed the factory reset.