ckb-next/ckb-next #763: K70 Keyboard - switching to hardware profile crashes ckb

nmenon, 2021-11-19:
## General information
K70 Keyboard - switching to hardware profile crashes ckb

### Source
How and where did you get this program, e.g.:
  * Built manually using quickinstall script
Commit: 1a73a0ea1da4 (HEAD -> master, origin/master, origin/HEAD) Fix hacks for WL in handle_bragi_key_input

## Logs & crash reports
[ckb-next-report-1637289033.gz](https://github.com/ckb-next/ckb-next/files/7567417/ckb-next-report-1637289033.gz)

https://user-images.githubusercontent.com/22279/142554965-c67d4c36-8ef9-41b1-bedc-43f3404f6adb.mp4

[gdb-tracelog.txt](https://github.com/ckb-next/ckb-next/files/7567420/gdb-tracelog.txt)

## What is the program's current behaviour?

When we switch to h/w profile stored in the keyboard, ckb-next segfaults.

## What is the expected behaviour of the program?
When we switch to h/w profile stored in the keyboard, ckb-next loads the hardware profile and continues to work.

## Any other notes & comments?

Suspected bad commit introduced at 449f5cbe97affe63374996b8d629b46665287b7c

Bisect Logs:

v0.3.1 - good
v0.4.0 - good
v0.4.1 - works
v0.4.2 - works
v0.4.3 - crashes 
v0.4.4 - crashes
git bisect start
# good: [1555ca97a7e34e8b1e5b5c4d78b4db472bab7f03] ckb-next v0.4.2
git bisect good 1555ca97a7e34e8b1e5b5c4d78b4db472bab7f03
# bad: [43d86ce205c837d9bb722aa16dc4bf8bfb78211e] Release v0.4.3
git bisect bad 43d86ce205c837d9bb722aa16dc4bf8bfb78211e
# good: [1c3ca772a3f95760c6c7b2931d28dd9d03dc7ba9] Move monochrome tray option to settings
git bisect good 1c3ca772a3f95760c6c7b2931d28dd9d03dc7ba9
# good: [44686f686d133cca0b022949d55c914b2c171682] Fix builds without mviz
git bisect good 44686f686d133cca0b022949d55c914b2c171682
# good: [d7ec4f55d1bbadbb669a92a250dd20caf0d7043d] Add nightsword to FW table
git bisect good d7ec4f55d1bbadbb669a92a250dd20caf0d7043d
# good: [608c0e190e7e97b536ef06ea0170f9e0621381dd] Fix nightsword indentation in usb.c
git bisect good 608c0e190e7e97b536ef06ea0170f9e0621381dd
# bad: [655083f0e2441d6e7a80f701d2fa2ea4e34973e7] Fix Qt 5.5 builds
git bisect bad 655083f0e2441d6e7a80f701d2fa2ea4e34973e7
# bad: [de7d6468528aeb19d0cc01b2903ec20ae8b19699] Merge branch 'xcb-profile-switch-v3'
git bisect bad de7d6468528aeb19d0cc01b2903ec20ae8b19699
# bad: [449f5cbe97affe63374996b8d629b46665287b7c] Squashed commit of the following:
git bisect bad 449f5cbe97affe63374996b8d629b46665287b7c
# first bad commit: [449f5cbe97affe63374996b8d629b46665287b7c] Squashed commit of the following:

tatokis, 2021-11-19:
Looks like my previous message didn't go through.

Can you please archive the config directory (~/.config/ckb-next) and attach it to the issue in a reply?

nmenon, 2021-11-19:
in my testing, I did a rm -rvf ~/.config/ckb-next before running ckb-next the first time in each of the bisect test I did. and it does crash - attached the version that works for me (v0.4.2 based)
[ckb-next.tar.gz](https://github.com/ckb-next/ckb-next/files/7570922/ckb-next.tar.gz)
.

nmenon, 2021-11-19:
happened to check from my other system with K95 and happens on it  as well. report from k95:
[ckb-next-report-1637334679.gz](https://github.com/ckb-next/ckb-next/files/7571106/ckb-next-report-1637334679.gz)
.
[ckb-next-config-k95.tar.gz](https://github.com/ckb-next/ckb-next/files/7571128/ckb-next-config-k95.tar.gz)

```
nmenon@akankob:~/Src/opensource/ckb-next$ rm -rvf ~/.config/ckb-next/;gdb /usr/local/bin/ckb-next
removed '/home/nmenon/.config/ckb-next/ckb-next_backup_1637334635.conf'
removed '/home/nmenon/.config/ckb-next/ckb-next.conf'
removed directory '/home/nmenon/.config/ckb-next/'
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/local/bin/ckb-next...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/local/bin/ckb-next 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe97fd700 (LWP 12572)]
[New Thread 0x7fffdda9c700 (LWP 12573)]
[New Thread 0x7fffdd29b700 (LWP 12574)]
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/nmenon/.config/ckb-next/ckb-next.conf"
Looking for old settings in: "/home/nmenon/.config/ckb/ckb.conf"
Backing up settings to "/home/nmenon/.config/ckb-next/ckb-next_backup_1637334779.conf"
2 keys backed up.
[New Thread 0x7fffd7fff700 (LWP 12581)]
Scanning "/usr/local/libexec/ckb-next-animations/gradient"
Scanning "/usr/local/libexec/ckb-next-animations/heat"
Scanning "/usr/local/libexec/ckb-next-animations/invaders"
Scanning "/usr/local/libexec/ckb-next-animations/life"
Scanning "/usr/local/libexec/ckb-next-animations/mviz"
Scanning "/usr/local/libexec/ckb-next-animations/pinwheel"
Scanning "/usr/local/libexec/ckb-next-animations/pipe"
Scanning "/usr/local/libexec/ckb-next-animations/rain"
Scanning "/usr/local/libexec/ckb-next-animations/random"
Scanning "/usr/local/libexec/ckb-next-animations/ripple"
Scanning "/usr/local/libexec/ckb-next-animations/snake"
Scanning "/usr/local/libexec/ckb-next-animations/wave"
[New Thread 0x7fffd75c1700 (LWP 12594)]
[New Thread 0x7fffd6dc0700 (LWP 12595)]
[New Thread 0x7fffd5c61700 (LWP 12596)]
[New Thread 0x7fffd5460700 (LWP 12597)]
[New Thread 0x7fffd48f0700 (LWP 12598)]
Starting "/usr/local/libexec/ckb-next-animations/wave"
Starting "/usr/local/libexec/ckb-next-animations/gradient"
Starting "/usr/local/libexec/ckb-next-animations/wave"
[New Thread 0x7fffb9cb6700 (LWP 12603)]
[New Thread 0x7fffb94b5700 (LWP 12604)]
Creating PulseAudio context
[Thread 0x7fffb9cb6700 (LWP 12603) exited]
[New Thread 0x7fffb9cb6700 (LWP 12605)]
Downloaded new firmware list. 39 entries found.

Thread 1 "ckb-next" received signal SIGSEGV, Segmentation fault.
0x00005555555ee0e6 in Kb::setCurrentMode(KbMode*) ()
(gdb) bt
#0  0x00005555555ee0e6 in Kb::setCurrentMode(KbMode*) ()
#1  0x00005555555eeba2 in Kb::setCurrentProfile(KbProfile*) ()
[... 9 lines cut ...]
#11 0x00007ffff721f269 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff545758c in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff711a815 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007ffff712265f in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x00007ffff54578a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00007ffff7121632 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff7173e95 in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#18 0x00007ffff71767ca in  () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff711a83c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff7122104 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff54578a8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007ffff5a195eb in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#23 0x00007ffff5a1b0b5 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#24 0x00007ffff59f233b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#25 0x00007fffec812260 in  () at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
[... cut ...]
#28 0x00007ffff20f67fc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007ffff54b085f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#30 0x00007ffff54558da in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007ffff545e984 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x0000555555592d4a in main ()
```

tatokis, 2021-11-19:
I can't get it to crash...
`ckb-next 0.4.4-145-g1a73a0e`

[image]

Can you build with `-DCMAKE_BUILD_TYPE=Debug` and then attach a `bt full` from gdb? Maybe that can help me find the cause without being able to reproduce it myself.

nmenon, 2021-11-19:
The following is with my K95:

```
nmenon@akankob:~/Src/opensource/ckb-next$ rm -rvf ~/.config/ckb-next/;gdb /usr/local/bin/ckb-next
removed '/home/nmenon/.config/ckb-next/ckb-next_backup_1637334779.conf'
removed '/home/nmenon/.config/ckb-next/ckb-next.conf'
removed directory '/home/nmenon/.config/ckb-next/'
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/local/bin/ckb-next...done.
(gdb) run
Starting program: /usr/local/bin/ckb-next 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe97fd700 (LWP 17157)]
[New Thread 0x7fffdda9c700 (LWP 17158)]
[New Thread 0x7fffdd29b700 (LWP 17159)]
QObject::startTimer: Timers cannot have negative intervals
Path to settings: "/home/nmenon/.config/ckb-next/ckb-next.conf"
Looking for old settings in: "/home/nmenon/.config/ckb/ckb.conf"
Backing up settings to "/home/nmenon/.config/ckb-next/ckb-next_backup_1637335943.conf"
2 keys backed up.
[New Thread 0x7fffd7fff700 (LWP 17161)]
Scanning "/usr/local/libexec/ckb-next-animations/gradient"
Scanning "/usr/local/libexec/ckb-next-animations/heat"
Scanning "/usr/local/libexec/ckb-next-animations/invaders"
Scanning "/usr/local/libexec/ckb-next-animations/life"
Scanning "/usr/local/libexec/ckb-next-animations/mviz"
Scanning "/usr/local/libexec/ckb-next-animations/pinwheel"
Scanning "/usr/local/libexec/ckb-next-animations/pipe"
Scanning "/usr/local/libexec/ckb-next-animations/rain"
Scanning "/usr/local/libexec/ckb-next-animations/random"
Scanning "/usr/local/libexec/ckb-next-animations/ripple"
Scanning "/usr/local/libexec/ckb-next-animations/snake"
Scanning "/usr/local/libexec/ckb-next-animations/wave"
[New Thread 0x7fffd75c1700 (LWP 17174)]
[New Thread 0x7fffd6dc0700 (LWP 17175)]
[New Thread 0x7fffd5c61700 (LWP 17176)]
[New Thread 0x7fffd5460700 (LWP 17177)]
[New Thread 0x7fffd48f0700 (LWP 17178)]
Starting "/usr/local/libexec/ckb-next-animations/wave"
Starting "/usr/local/libexec/ckb-next-animations/gradient"
Starting "/usr/local/libexec/ckb-next-animations/wave"
[New Thread 0x7fffb9cb6700 (LWP 17183)]
[New Thread 0x7fffb94b5700 (LWP 17184)]
Creating PulseAudio context
[Thread 0x7fffb9cb6700 (LWP 17183) exited]
[New Thread 0x7fffb9cb6700 (LWP 17185)]
Downloaded new firmware list. 39 entries found.

Thread 1 "ckb-next" received signal SIGSEGV, Segmentation fault.
0x00005555555c5f58 in KbMode::light (this=0x0) at /home/nmenon/Src/opensource/ckb-next/src/gui/kbmode.h:62
62	    inline KbLight* light() { return _light; }
(gdb) bt full
#0  0x00005555555c5f58 in KbMode::light() (this=0x0) at /home/nmenon/Src/opensource/ckb-next/src/gui/kbmode.h:62
#1  0x00005555555f5327 in Kb::setCurrentMode(KbMode*) (this=0x555555d60100, mode=0x0) at /home/nmenon/Src/opensource/ckb-next/src/gui/kb.cpp:839
#2  0x00005555555f5253 in Kb::setCurrentProfile(KbProfile*) (this=0x555555d60100, profile=0x7fffb00189a0) at /home/nmenon/Src/opensource/ckb-next/src/gui/kb.cpp:832
#3  0x0000555555625d87 in KbWidget::on_profileBox_activated(int) (this=0x555555e47ee0, index=1) at /home/nmenon/Src/opensource/ckb-next/src/gui/kbwidget.cpp:204
[... 22 lines cut ...]
#23 0x00007ffff5a195eb in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#24 0x00007ffff5a1b0b5 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) ()
    at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#25 0x00007ffff59f233b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5
#26 0x00007fffec812260 in  () at /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#27 0x00007ffff20f6537 in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007ffff20f6770 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007ffff20f67fc in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#30 0x00007ffff54b085f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007ffff54558da in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00007ffff545e984 in QCoreApplication::exec() () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#33 0x00005555556673ae in main(int, char**) (argc=1, argv=0x7fffffffd688) at /home/nmenon/Src/opensource/ckb-next/src/gui/main.cpp:370
        tmpSettings = 0x0
        a = <incomplete type>
        translator = <incomplete type>
        currentSettingsVersion = 0
        parser = {d = 0x555555bc3400}
        errorMessage = {static null = {<No data fields>}, d = 0x7ffff55455a0 <QArrayData::shared_null>}
[... cut ...]  = {<QListSpecialMethods<KbWidget*>> = {<No data fields>}, {p = {static shared_null = {ref = {atomic = {_q_value = {<std::__atomic_base<int>> = {static _S_alignment = 4, _M_i = -1}, <No data fields>}}}, alloc = 0, begin = 0, end = 0, array = {0x0}}, d = 0x555555a4d7a0}, d = 0x555555a4d7a0}}, restoreAction = 0x555555c570b0, closeAction = 0x555555c577c0, trayIconMenu = 0x555555c58240, trayIcon = 0x555555c63b90, ui = 0x555555a0b1d0, sigNotifier = 0x555555e5ebe0, updater = 0x555555e48dd0}
(gdb) 
```

nmenon, 2021-11-19:
@tatokis -> could you backup and remove ~/.config/ckb-next and try running ckb-next and switch to hw saved profile? I just was trying to step through with ddd with the debug build and noticed (surprise for me) it did'nt crash.. maybe this is happening the very first time perhaps? that might help a clue?

tatokis, 2021-11-19:
I did do that and it didn't crash for me.
It looks like it's a bug in the GUI that is only happening with the version of Qt on 18.04.

I just tested it on my Xubuntu 18.04 32 bit machine and it crashed there as well.
I think it has to do with when Qt dispatches the signal for switching to the profile. It's probably a really bad race condition.

tatokis, 2021-11-19:
I guess I did it wrong. Sure enough, removing the config makes it crash now. This is a really nasty bug.
I'll cook up a fix for it.

tatokis, 2021-11-19:
I was hoping for a proper solution, but the whole thing is unfortunately really spaghetti. It should be fixed now at least.

closed 2021-11-19 by tatokis

nmenon, 2021-11-19:
@tatokis thanks a bunch. hey, evolution of code, but it does work.. and is an awesome stuff you have put together. Thank you for doing this and keeping this alive - I do know that i would be working slower if my macros and nifty hooks are'nt around..!!!

commit 1c98bf9b (2021-11-19): Fix GUI segfault on HW profile switch w/clean config

Closes #763