ckb-next/ckb-next #765: Corsair K100

GermanKingYT, 2021-11-27:
# Before you proceed

- Check if your device is listed in the [wiki](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware).
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

---

Enter the name of your device here and attach the report generated above.

GermanKingYT, 2021-11-27:
[ckb-next-report-1638029240.gz](https://github.com/ckb-next/ckb-next/files/7612011/ckb-next-report-1638029240.gz)

bG9sIHVyIGhvbW8, 2021-12-02:
#651

closed 2021-12-03 by tatokis