ckb-next/ckb-next #766: Manage Events Adjustment

MoonBurst, 2021-11-27:
## What is the program's current behaviour?
Currently you right click the profile name, click manage events, and it pops out a menu.
## What would you like it to do instead?
Similar to Lighting, Binding, Performance, and Device, it being in line. ALSO, a text piece to explain "Type xprop or xwininfo in terminal and click a window to find it's name". As would make understanding how to use it easier to the end user.
## Any other notes & comments?
It's a great feature, and I'm a big fan of the program. Thank you for making this program.

filbert14, 2022-10-08:
Hi! Is this project/issue still open? I'd like to contribute and would appreciate if I were given some pointers on where to start :)

MoonBurst, 2022-10-08:
> Hi! Is this project/issue still open? I'd like to contribute and would appreciate if I were given some pointers on where to start :)

Well, I mean, I'm not a strong programmer and so haven't done it. But it were something that occurred to me and hasn't been done yet as far as I know.

tatokis, 2022-10-08:
Yes, this issue is still open. It should just be a matter of opening `kbmodeeventmgr.ui` in Qt Creator and adding text at an appropriate place.

filbert14, 2022-10-08:
All clear, I will immediately start working on this :)

filbert14, 2022-10-08:
Hi! I opened a pull request - did not make it inline though, as I'm not sure if changes to that are desired based on your reply.

commit c7d04173 (2022-10-13): Add xprop tip in window event manager

Closes #766
closed 2022-10-13 by tatokis