ckb-next/ckb-next #767: Cannot write to hardware, used to be able to on Fedora 34.

ThoriumTextile, 2021-11-28:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).  _[ Done ]_
- If possible, try the latest code by building the software manually.  _[ Not tested/tried ]_
- File unrelated issues separately. _[ N/A ]_

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.** _[ Done ]_

## General information

### Source
How and where did you get this program, e.g.:
From OpenSUSE's official repos.

## Logs & crash reports
[ckb-next-report-1638101711.gz](https://github.com/ckb-next/ckb-next/files/7613511/ckb-next-report-1638101711.gz)

## What is the program's current behaviour?
Works as expected but not the hardware save function.

## What is the expected behaviour of the program?
Saving to hardware after making changes and choosing so.

## Any other notes & comments?
**Kernel:** 5.15.2-1-default #1 SMP Sat Nov 13 14:10:12 UTC 2021 (26a203b) x86_64 x86_64 x86_64 GNU/Linux
**OS:** OpenSUSE Tumbleweed
**Hardware:** Corsair Harpoon RGB Pro

tatokis, 2021-11-28:
Did it actually work before?
The button may have been visible, but the way hardware profiles worked was different to how ckb-next handled them, so clicking it wouldn't have done anything at all.

ThoriumTextile, 2021-11-30:
> Did it actually work before? The button may have been visible, but the way hardware profiles worked was different to how ckb-next handled them, so clicking it wouldn't have done anything at all.

Yes. Once i turned off my computers, my profiles were there and still functional.

5ch4um1, 2022-05-03:
This never worked for me. Clicking the "save to hardware" button only seems to apply the changes, as in, sending them to the device. The button is greyed out unless you make some changes, and it is greyed out as well if that "use dpi indicator" checkbox is ticked.

And by the way, the lpc11u37 seems to be supported by mbed, has anybody ever thought about maybe writing a custom firmware for this pointy fishing spear with colours mouse thingy? This would in theory allow to bake those settings and macros into the firmware and even have them in bios mode? There also should be 4k of eeprom on that chip, so it might be possible to add that "save to hardware" functionality for real?
And for the stock firmware, has anybody looked into maybe changing the default bios mode colour?
Like, don't get me wrong, i love this mouse, but the led doesn't really seem that useful when my hand is over it and i can't see it, so i would generally like to just turn it off.
Or maybe desolder it and put it more to the front, where it could do something useful. Or replace it with a few ws2812 and free up 2 more gpio pins?

ThoriumTextile, 2023-03-12:
@tatokis Any update on this? This still seems to be broken.

Ravenslofty, 2023-03-12:
Please attach a ckb-next report; there are about a million different things that could be going wrong and there's no particular guarantee you have the same issues.

ThoriumTextile, 2023-03-12:
> Please attach a ckb-next report; there are about a million different things that could be going wrong and there's no particular guarantee you have the same issues.

I already did, but good thing i'm on NixOS now.
Nonetheless the issue still persists.
[ckb-next-report-1678625965.gz](https://github.com/ckb-next/ckb-next/files/10950854/ckb-next-report-1678625965.gz)

tatokis, 2023-03-12:
I'm sorry, this makes no sense. You say "profiles", but as far as I know ckb-next could never write multiple profiles to a mouse as it's a different part of the protocol, so it simply wouldn't have done anything. Only other thing I can think of is that there was a firmware update that stopped accepting the old commands, as they weren't suited for multiple profiles.

Is it possible that you misunderstand what "Save to hardware" does? Can you exactly describe what the behaviour before and after is?

tatokis, 2023-03-15:
Okay, it looks like your mouse only supports a single profile. I must have gotten confused with something else. Apologies.
Please open up usb.h, and remove `|| (kb)->product == P_HARPOON_PRO` from USES_FILE_HWSAVE.

Then, run the daemon in a terminal, and click the save to hardware button.
Send the whole output of the daemon from that run here.

ThoriumTextile, 2023-03-15:
> Okay, it looks like your mouse only supports a single profile. I must have gotten confused with something else. Apologies. Please open up usb.h, and remove `|| (kb)->product == P_HARPOON_PRO` from [...]

I'm not quite sure where i'm supposed to find usb.h.

tatokis, 2023-03-17:
https://github.com/ckb-next/ckb-next/blob/bc6e222e234d7bb03b89af1e6894e799b6b0d224/src/daemon/usb.h#L264

ThoriumTextile, 2023-03-17:
> https://github.com/ckb-next/ckb-next/blob/bc6e222e234d7bb03b89af1e6894e799b6b0d224/src/daemon/usb.h#L264

Am I supposed to recompile ckb-next with said line?

closed 2023-06-08 by ThoriumTextile