ckb-next/ckb-next #769: Suport for the uncomon K83

M3t4ll1k0, 2021-11-29:
Just bough this keyboard for a TV conected pc nvidia shield and LG tv
Being a keyboard i assumed i was plug and play, to be hones not much info about it, so i dont know what feature can be used in linux. That assuming if i boot a win partition, some configuration will be stored in the hardware.
But i did not received it yet and seem it does not work like a HID device, so Im wondering if even if need software like this to even work or some features hare managed by windows iCUE like macros and such.
Would love to this protect support this keyboard. Thanks.

M3t4ll1k0, 2021-12-01:
Just received the K83 today, the gui app does not recognize it, nor allow to configure gestures touch-pad speed... etc. Could not care less of fucking colors (is only white backlit).
Will try the daemon quirks etc...
I dont code but if there is soothing i can help with devs/maintainers can contact me on alejandro.buitron.corral@gmail.com

tatokis, 2021-12-01:
There is no support for it, as you just found out. If we do add support for it, it would probably be just for macros and lighting. I don't really feel like implementing gestures, nor do I know how or have the time to find out, and even just getting the trackpad to work through this in the first place will probably be a pain for what is only a single device.

Only way this would be implemented easily would be if the gestures were handled by the hardware, which would fire off an event to the software. I don't really know how it works. Someone needs to look into it.

You might have some luck finding generic touchpad gesture software that works with libinput, or something along those lines.

For anything else, patches welcome, basically.

M3t4ll1k0, 2021-12-01:
Yea, totally understand and agree with you. Funny shit, macros are not
stored in hardware, but it seem other functions like gesture, speed
acceleration, and a lot more customization features that appear on some at
least are stored in the hardware, sin windows iCUE says harware store
profiles. The fucked up is that iCUE say that at this moment there are
features not suported, that will fix it... so besides macros and remaping
wich are not stored in the hardware, besides the fucking backlight, alll of
that customization including gestures lick pinch i want off and the speed
wich is ridiculous, barelly un usable... but ubuntu detects it as a mouse
so i can even chanche it, and to activate normal scrolling and no natural I
have to do the oposite... . So yea totally right, my best change is a
generic trackpad tool, know any btw?
Before purging your packages and repo, by any chance the lighting
connection is managed by your deamon or other piece of code, to see if cna
do something with your documentation.

closed 2024-05-02 by gravityfargo