ckb-next/ckb-next #771: quickstart failing to install on arm64 [jetson-nano, which is similar to a raspberry pi]

nubonics, 2021-12-02:
## General information

### Source
How and where did you get this program, e.g.:
  * Built manually using quickinstall script [ this repo ]

## Logs & crash reports
nubonix@nubonix-jetson:/opt/ckb-next$ ./quickinstall 
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- ckb-next version: 0.4.4-146-g1c98bf9 (Non-release)
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST - Failed
-- Found Iconv: /usr/lib/aarch64-linux-gnu/libc.so
-- Checking for module 'libudev'
--   Found libudev, version 237
-- Found UDEV: /lib/aarch64-linux-gnu/libudev.so  
-- Searching for running ckb and/or ckb-next GUI
-- NOTE:
`
    Privileged access is required for operations upon the daemon at
    configure time as it is owned by root. If this is unacceptable,
    disable SAFE_INSTALL and prepare the filesystem manually.
-- systemd detected
-- Generating and importing ckb-next-daemon.service (systemd)
inactive
Failed to get unit file state for ckb-daemon.service: No such file or directory
inactive
Failed to get unit file state for ckb-next-daemon.service: No such file or directory
-- Found QuaZip: /usr/lib/aarch64-linux-gnu/libquazip5.so  
-- Found PulseAudioSimple: /usr/lib/aarch64-linux-gnu/libpulse-simple.so  
-- Found PulseAudio: /usr/lib/aarch64-linux-gnu/libpulse.so  
-- Found ZLIB: /usr/lib/aarch64-linux-gnu/libz.so (found version "1.2.11") 
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so
-- Looking for XOpenDisplay in /usr/lib/aarch64-linux-gnu/libX11.so;/usr/lib/aarch64-linux-gnu/libXext.so - found
-- Looking for gethostbyname
-- Looking for gethostbyname - found
-- Looking for connect
-- Looking for connect - found
-- Looking for remove
-- Looking for remove - found
-- Looking for shmat
-- Looking for shmat - found
-- Looking for IceConnectionNumber in ICE
-- Looking for IceConnectionNumber in ICE - found
-- Found X11: /usr/lib/aarch64-linux-gnu/libX11.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'xcb'
--   Found xcb, version 1.13
-- Checking for module 'xcb-ewmh'
--   Found xcb-ewmh, version 0.4.1
-- Checking for module 'xcb-screensaver'
--   No package 'xcb-screensaver' found
CMake Error at src/gui/CMakeLists.txt:274 (message):
  XCB-Screensaver or Qt5 X11Extras was not found.  Either install them or
  pass -DUSE_XCB=0 to disable idle detection.
`

-- Configuring incomplete, errors occurred!
See also "/opt/ckb-next/build/CMakeFiles/CMakeOutput.log".
See also "/opt/ckb-next/build/CMakeFiles/CMakeError.log".
An error occurred,
press enter to exit.

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behaviour?
Will not install

## What is the expected behaviour of the program?
For the installation of quickstart to install

## Any other notes & comments?
system: nvidia's jetson nano
architexture: arm64
operating system: nvidia's jetson nano OS which was built on top of ubuntu

tatokis, 2021-12-02:
Please follow the installation instructions in the wiki for ubuntu. If you can't add the PPA and install ckb-next that way, then please install the dependencies as indicated.
https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#recommended-packages
https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#build-from-source

nubonics, 2021-12-02:
Awesome! Didnt know there was a ppa,
Thanks!

closed 2021-12-02 by nubonics