ckb-next/ckb-next #773: systemd integration

rrediske, 2021-12-07:
## What is the program's current behaviour?
Using the --enable-experimental flag requires manually editing the systemd ckb-next-daemon.service file. This file is overwritten any time there are updates or rebuilds, requiring it to be manually edited afterward.

## What would you like it to do instead?
Systemd allows a method for overriding defaults via ```systemctl edit ckb-next-daemon```, but from my understanding, that method does not pass run time flags like --enable-experimental. After googling around a bit, there are some work arounds, but they seem overly complex like creating a text file with command arguments.

Is it possible to create an entry under the ```[Service]``` heading in the systemd unit file, for example, ```EnableExperimental=yes```? That way, systemd can create an override file that survives updates and rebuilds?

## Any other notes & comments?
I hope this is possible and doesn't require hours of code. It's not a great pain to edit the unit file manually, but it's a nice quality of life experience and seems to follow conventions a little better.

Technetium1, 2022-01-07:
It's not too bad! Changes that are [done this way](https://www.digitalocean.com/community/tutorials/how-to-use-systemctl-to-manage-systemd-services-and-units#:~:text=started%20or%20enabled.-,Editing%20Unit%20Files,-While%20the%20specific) should dynamically apply to the service even when it's updated.

tatokis, 2022-01-07:
I don't see a reason for systemd integration. Just as you can edit the unit to enable experimental (as you propose), you can add the argument, minus maybe some PATH issues.

I wrote up a quick way to do it in this post https://github.com/ckb-next/ckb-next/issues/543#issuecomment-1007353508 however I do think I need to make an entry in the wiki with the appropriate disclaimers, and the actions to take if things go wrong.
In general, since it can leave someone keyboardless, I am not comfortable with making it really simple for people to enable. It's much easier to stop the daemon if your mouse isn't working, than if your keyboard isn't.

ReillyBrogan, 2023-08-10:
A better approach would be to add `EnvironmentFile=-/etc/sysconfig/ckb-next-daemon` to the package-shipped service file and then change `ExecStart=/usr/bin/ckb-next-daemon` to `ExecStart=/usr/bin/ckb-next-daemon $CKB_OPTIONS`. 

This would allow an end-user to create `/etc/syconfig/ckb-next-daemon` and add `CKB_OPTIONS=--enable-experimental` to it, or any other flags that may be useful in the future. Then they don't need to muck about with systemd override files (which are non-intuitive with how you have to set the `ExecStart=` before you can actually override it). 

The documentation for end-users on this would be just:
```
sudo mkdir -p /etc/sysconfig
echo "CKB_OPTIONS=--enable-experimental" | sudo tee /etc/sysconfig/ckb-next-daemon
sudo systemctl restart ckb-next-daemon
```