ckb-next/ckb-next #774: Make the daemon support full functionality even with the the GUI app closed

ecruz1986, 2021-12-17:
I'm on Arch Linux.

## What is the program's current behaviour?
Today, we need to keep the ckb-next GUI open to enable full functionality of the ckb-next app, like the mute button, caps lock, num lock key color based on their state. If we close the GUI app, those functionalities stop working, even though the daemon is running.

## What would you like it to do instead?
The daemon should take care of those functionalities even when the GUI is closed. The GUI should only be used occasionally to interface with the daemon when needed, it should not need to be started on every login. These background operations traditionally should be the daemon's responsibility.

## Any other notes & comments?
Today the current profile is stored per user in their home folder. With the new  proposed architecture, the concept of an active "system wide" profile should come to exist, and that is what should be read and used by the daemon. Users with admin privileges should be allowed in the GUI to save their profile into the "system wide" profile, and that would be used by the daemon running as root. The profile would be activated as soon as the daemon starts running on boot, not even requiring a user login session.

fleischie, 2021-12-17:
Back when I first started working on a command line interface (which I think never manifested in the way I proposed) I had a similar question. But as far as I understand the daemon is merely for administering a communication channel between the devices and the client application.
The GUI on the other hand is storing the settings/animations/etc. and continuously sending that to the device to make it light up, resp. waiting for keypresses to operate on macros.

I think there is the "-b" switch, that runs the client app in the background. Or more specifically hides the UI.

On the other hand there is the Wiki page https://github.com/ckb-next/ckb-next/wiki/CKB-Daemon-Manual that outlines a direct communication to the daemon without the GUI, but that would require you to take care of storing of and operating on your profile manually. Which you could use to set up a program that is run with your init system, and which I think does what you ask for. (Of course in your responsibility and not in the project's core.)

ecruz1986, 2021-12-17:
Hi, thanks for taking your time to look into this!

I'm not really interested into just fixing it for myself. The goal of this is really to improve this (already great) software so that all of its users can benefit from a better out-of-the box experience.

I understand that today the daemon is just for communicating between devices and clients. And I'm proposing a paradigm shift from that. It could become more intelligent as to provide all the functionality without requiring a GUI or any other type of client application to be running.

Ravenslofty, 2022-01-20:
This is probably not going to happen.

As the daemon is running as root, the less code it contains, the better.

closed 2022-04-04 by tatokis