ckb-next/ckb-next #775: K55 RGB PRO XT

cerias, 2021-12-25:
# Device Report 
```
ckb-next-dev-detect - Sat Dec 25 10:41:31 CET 2021

Kernel: Linux niflheim 5.11.0-41-generic #45-Ubuntu SMP Fri Nov 5 11:37:01 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
Distributor ID:	Ubuntu
Description:	Ubuntu 21.04
Release:	21.04
Codename:	hirsute

USB Devices:

Bus 001 Device 003: ID 1b1c:1ba1 Corsair CORSAIR K55 RGB PRO XT Gaming Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1ba1 
  bcdDevice            1.03
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR K55 RGB PRO XT Gaming Keyboard
  iSerial                 3 F5001BC55F85C1DEAF7BA4A70500F031
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      82
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
[... 244 lines cut ...]
[E] os_usb_interrupt_out (via bragi_common.c:9): No such file or directory, res = 0xffffffff
[F] start_bragi_common (device_bragi.c:111): ckb1: Couldn't get bragi device mode. Aborting
```

While i have no idea and it was ask on an other thread here are the id descriptors:

```
001:003:002:DESCRIPTOR         1640427271.081253
 06 42 FF 09 02 A1 01 15 00 26 FF 00 75 08 95 40
 09 02 81 02 C0

001:003:001:DESCRIPTOR         1640427271.083250
 06 42 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

001:003:000:DESCRIPTOR         1640427271.086224
 05 01 09 06 A1 01 85 01 05 08 95 03 75 01 19 01
 29 03 15 00 25 01 91 02 95 05 91 03 05 07 75 01
 19 E0 29 E7 95 08 81 02 19 00 29 67 95 68 81 02
 19 87 29 8E 95 08 81 02 C0 05 0C 09 01 A1 01 85
 02 95 01 75 10 15 00 26 FF 03 19 00 2A FF 03 81
 00 C0
 ```

I would happy to support the implimentation with time and a few € if it's okay.

hlechner, 2022-08-04:
@tatokis on issue #838
> There's a branch you can test out https://github.com/ckb-next/ckb-next/tree/k55pro. It needs some changes before it can be merged, but it should be functional.

Thanks for working on that! I know the **K55 PRO** and **K55 PRO XT** are **not** the same model, the main difference is the individual RGB per key on XT model.

But I tried it just to see if something worked: just changing the product id (from **0x1ba4** to **0x1ba1** on **src/daemon/usb.h**)

The biding seems to work fine (no problem found) and as expected the lighting do not work (probably due to individual keys instead of zones)

it should be no surprise that it doesn't work since is different model, but if helps here what happens using it on XT:
The keys: A,B,C,D,E,F,G,H,I,J,K turn red and the rest just keep off, changing the animation or color do not change anything on keyboard

If the XT model will be supported, I can help test since I do have the hardware

tatokis, 2022-08-04:
@hlechner Give this a try https://github.com/ckb-next/ckb-next/compare/k55proxt

tatokis, 2022-08-04:
I found a mistake and had to force push. If you downloaded the code already, please redownload it (or `git reset --hard HEAD~1` and then `git pull` again).

hlechner, 2022-08-04:
@tatokis thats awesome! it's working:

[image]

Now that its working I will manage to learn how to use the software so I can report any bug

marmotz, 2022-08-04:
I have this keyboard too and i can confirm it's working !
Thanks !

commit 2bd1a315 (2022-08-04): Merge branch 'k55proxt'

Closes #775
closed 2022-08-04 by tatokis