ckb-next/ckb-next #782: New release please?

BloodyIron, 2022-01-09:
With the changes that have happened since the March 2021 release, can we please see a new release come out so that the updates can eventually roll out to us using repos? :)

BloodyIron, 2022-03-26:
@tatokis ?

tatokis, 2022-04-04:
If you want to speed up the release, you can start by building the latest code from git, fixing the warnings in the daemon and making PRs.

It's basically only me here, and I don't presently consider it acceptable to make a release when there are basic features that need to be implemented.
The daemon can't read the firmware version from new devices. There's no support for pollrates <1ms. The daemon gets stuck in an endless loop if it can't talk to bragi devices. It's not been decided what to do with experimental devices on release. And so on, and so on.

tatokis, 2022-04-18:
As a sidenote, I've created a milestone. I'll try to get the release out in the upcoming weeks, but I can't promise anything.
https://github.com/ckb-next/ckb-next/milestone/3

closed 2022-05-27 by tatokis

BloodyIron, 2022-05-28:
Yay! Hit my repos, thanks again! :D