ckb-next/ckb-next #785: CORSAIR VIRTUOSO SE Wireless Gaming Headset

david-waterworth, 2022-01-20:
I've attached a report for my `CORSAIR VIRTUOSO SE Wireless Gaming Headset`, it's not detected by `ckb-next` (Ubuntu 20.04 installed via apt) and as far as I can see it's not on the list of supported devices. Would be great if I could just turn off the LED's from linux

[ckb-next-report-1642649626.gz](https://github.com/ckb-next/ckb-next/files/7902080/ckb-next-report-1642649626.gz)

Arto-O, 2022-01-22:
I have the same issue. I'm looking to switch to Linux as my daily driver and would like to be able to control my Virtuoso headphones' RGB.

RichardsonWTR, 2022-02-09:
Have you guys experienced poor battery life on Linux? Just got brand new and after a full charge the battery died in just 4 hours!

Arto-O, 2022-02-09:
> Have you guys experienced poor battery life on Linux? Just got brand new and after a full charge the battery died in just 4 hours!

I switched to Manjaro on my laptop around a month ago and haven't noticed any difference in battery life compared to Windows. They can usually last me full eight hours of listening to music.

It seems that the headphones need iCUE to give verbal warnings of low battery. On Linux they just beep multiple times. A reason you might think the headphones died is this annoying feature/bug on both OSs where the headphones just turn off if you sit still for too long. When this happens, I shake my head around like a crazy person and that turns the headphones back on.

To see if the battery is actually low, look at the battery indicator on the left cup. It goes from green to yellow and then to red. Blinking red means it lost connection with the dongle. Of course, if the headphones turned off, you need to get them back on to see the battery LED.

Arto-O, 2022-02-15:
The "hibernate" feature definitely works without iCUE for me.

david-waterworth, 2022-03-06:
I find the hibernate function extremely unreliable, they turn off, then turn back on again randomly through the night - to the point that they're often flat when I come to use them. It's not linux/windows though, it just seems to be a flaw in the sensor that detects if they're in use or not.

RichardsonWTR, 2022-05-02:
To sum up my experience until today and my workarounds to the battery problem: 
* The battery life is very short when all the lights are on, at least in Linux. 
* The hibernate sensor is kinda too sensible. Small shakes on the table can wake up the headset sometimes.

My solutions to the battery problem on Linux
* Disable the lights (on Windows, of course)
* Adjust the sleeping time of the headphone - the standard time is too long (10 minutes) - on Windows, too.
* On Windows I can disable the green light on the microphone when it's on. Unfortunately this setting only works on Windows. 

Hope it helps

Grocel, 2022-06-01:
I believe the hibernate feature is controlled on hardware/firmware level inside the headset. PC driver software is not needed for it and also does not control it. However iCUE can enable or disable this feature and also change the time until sleep. 

Both attributes are saved on device.

[image]

Setting called `Schlafmodus` in the German UI controls these hibernation attributes.

gravityfargo, 2024-05-02:
When you ran the report, was the headset plugged in or just wireless?