ckb-next/ckb-next #786: Ironclaw RGB not able to rebind profile cycle button

Srdjan-V, 2022-01-26:
## General information

### Source
How and where did you get this program, e.g.:
  * Package from AUR
  * ckb-next-git

## Logs & crash reports
[ckb-next-report-1643159095.gz](https://github.com/ckb-next/ckb-next/files/7938256/ckb-next-report-1643159095.gz)

## What is the program's current behaviour?
After rebinding the profile cycle button to middle mouse button or print screen (maybe others) it simply does nothing.

## Any other notes & comments?
I have tried this in virtualbox through passthrough, but this bug also existed about 6 months ago when i tried linux on my pc.

jdmj, 2022-02-11:
Same for the Nightsword. Weirdly, one function "Profile Up" button can be remapped to is brightness control.

tatokis, 2022-04-04:
Should be fixed in ckb-next-git.

commit 19a25c5c (2022-04-04): Fix profswitch rebinding in mice

There are two profswitch entries in the keymap, one for keyboards and
one for mice. This commit adds code to dynamically patch in the mouse
profswitch, disabling the keyboard one, if a mouse is detected.

Closes #786
closed 2022-04-04 by tatokis